zephyrproject-rtos/zephyr #85965: Twister support for external J-Link devices

LimeSlice, 2025-02-18:
**Is your enhancement proposal related to a problem? Please describe.**

I am currently running into an issue where I cannot determine how to get Twister to flash a custom board connected with an external SEGGER J-Link. When generating a hardware map, Twister pulls the symlinks from `/dev/serial/by-id` directory, where an external J-Link does not show. If you run `lsusb`, you can see the J-Link is connected. 

The documentation mentioned [here](https://docs.zephyrproject.org/latest/develop/test/twister.html#executing-tests-on-multiple-devices) mentions the use of `--device-serial-pty` to provide custom flash functionality, yet there is no sample that exists to provide information on how the script needs to be formatted.

**Describe the solution you'd like**

I am able to flash my custom board (SoC is an nRF5340) with the nrfjprog runner using: `west flash --snr <jlink-serial-nb> --recover` without any issues. I would expect a similar option to be available to me in Twister, but cannot get this due to the J-Link not being a serial device.

Either a better solution for external J-Link devices should be provided and/or an example for how to write the script for `--device-serial-pty` should be written.

**Describe alternatives you've considered**

I've tried to write a `--device-serial-pty` script called "script.sh" that simply contains:
```
west flash --recover
```
while executing my device tests with:
```
twister -p CUSTOM_BOARD -T app -vv --device-testing --west-flash=script.sh --device-serial-pty=script.sh
```
This results in my tests executing without the device being flashed. This works if the device is pre-flashed, but is not the desired behavior.

**Additional context**

My board is using bluetooth. The nRF5340 has two CPU cores for this, which requires me to use `sysbuild`, thus also requiring me to use `--west-flash`. This has caused additional complications, though not the root of my issue.

nordicjm, 2025-02-19:
There are lots of examples including one for this exact soc on https://docs.zephyrproject.org/latest/develop/test/twister.html#using-single-board-for-multiple-variants

nordicjm, 2025-02-19:
> [@nordicjm](https://github.com/nordicjm) I took a look at what you suggested, but did not find this helpful. You pointed me to an example that may be the same soc, but is a dev kit that has an on-chip JLink that allows the dev kit to be detected via serial port. I am using a custom board that cannot be detected as a serial device.
> 
> I did however look at the section above: https://docs.zephyrproject.org/latest/develop/test/twister.html#overriding-board-identifier, yet still was unable to successfully flash my board via Twsiter.
> 
> I first ran `twister --generate-hardware-map`. Due to the device being an external JLink probe, the device was unlisted here. I manually inserted the JLink as:
> 
> ```
> - connected: true
>   id: '852002481'
>   probe_id: '852002481'
>   platform: CUSTOM_BOARD
>   product: J-Link
>   runner: nrfjrpog
>   serial: null
> ```
> 
> then attempted to run twister with:
> 
> ```
> twister -T app -vv --device-testing --west-flash --hardware-map map.yaml
> ```
> 
> This resulted in the same result where I was unable to flash the device due to the hardware not being detected.
> 
> ```
> DEBUG   - Finished building <path>/app for CUSTOM_BOARD in 3.73 seconds
> DEBUG   - Determine test cases for test instance: CUSTOM_BOARD/app.test
> DEBUG   - Determine test cases for test suite: app.test
> DEBUG   - Test instance CUSTOM_BOARD/app.test already has 1 cases.
> DEBUG   - run test: CUSTOM_BOARD/app.test
> DEBUG   - Reset instance status from 'passed' to None before run.
> ERROR   - No device to serve as CUSTOM_BOARD platform.
> ERROR   - Hardware is not available
> DEBUG   - run status: CUSTOM_BOARD/app.test failed
> INFO    - 1/1 CUSTOM_BOARD    app.test                                     FAILED Hardware is not available (device)
> ```
> 
> I additionally tested updating the runner to `jlink`, as well as trying to execute with `twister -T app -vv --device-testing --west-flash="device-id=852002481" --hardware-map map.yaml` with both variants.
> 
> Some additional info I realize I left out is I'm currently using Zephyr v3.7 with West v1.2 if this makes a difference.

Remove `probe_id` and set `serial` to the UART of the device. If you do not have a UART for the board then you cannot use twister to test

LimeSlice, 2025-02-19:
@nashif  How would I provide the UART? I only can find the JLink if I execute `lsusb`:
```
Bus 003 Device 032: ID 1366:1020 SEGGER J-Link
```
Is there a way to convert this to a path for the hardware map?

nordicjm, 2025-02-19:
Not sure what you mean how you provide the UART, you use something like an FTDI which gives you UART pins and connect it up to your board. If you have no UART transceiver to connect to a PC on the board, then I'm not really sure what it is exactly you're hoping to test, you can't test anything

LimeSlice, 2025-02-19:
My board is connected via a JLink Plus in order to flash the NRF5340 via the SWDCLK and SWDIO pins to the soc. I'm attempting to provide this so that Twister might flash the board prior to executing it's tests.

Additionally I have an FTDI cable connected to P0.17 and P0.18 on the NRF5340 which I have Kconfig options to enable/disable log output via UART. I am NOT trying to get the log output nor is it enabled on the build I'm intending to use.

I only want Twister to flash, then execute my tests. I am using a pytest harness with Twister that communicates with an external device in order to validate BLE functionality. A test will pass based on the pytests, whether the bytes returned via BLE are valid or not. But because I intend to have a Twister setup that might execute multiple tests, I need to be able to flash my board prior to running a Twister test with the intended firmware. 

Per my original comment, if I run with `--device-serial-pty`, I am able to get around the flashing + missing hardware and am able to make Twister run through my pytests. If the hardware was pre-flashed, I can successfully execute my pytests, but I need it to flash the soc prior running through my pytests.

LimeSlice, 2025-02-19:
Determined with the [Discord chat](https://discord.com/channels/720317445772017664/1340087557999824927) that my issue was that I was not using the Twister `dut` fixture within my pytest which is what triggers the scripts or flashing to occur.

I still stand by my original request for additional documentation to be created on `--device-serial-pty` and how it should be structured. Additional documentation to describe how it gets linked into Twister and a sample would be extremely beneficial.