zephyrproject-rtos/zephyr #86006: driver: Initialization of drivers at APPLICATION level

th-fischer2020, 2025-02-19:
**Describe the bug**

Currently, we are using zephyr v3.7.1 for our Application and the initialization of drivers on level `APPLICATION` is marked as deprecated. On later Versions e.g. v4.0.0 this will run into an compilation error because drivers cannot be initialized on level `APPLICATION`.  

Due to requirements of IO-Link communication we have to start communication within <300ms since power-up. The driver initialization of `display_st7735r` takes about >240ms because of necessary sleeps. Hence IO-Link specification cannot be guaranteed.  

Our current solution is, to initialize display driver, lvgl and UI on late `APPLICATION` level. By the upgrade to next zephy version our application will break because display driver cannot initialized on `APPLICATION` level.  
To push IO-Link doings into POST_KERNEL before display initialization, will cause an exception because of `z_init_static()`  has not be done at this Initialization Level. In addition pushing IO-Link doings to `POST_KERNEL` ist not the right way.

**To Reproduce**

**Expected behavior**

What is the reason for not allowing driver init on APPLICATION level? 
Is it an Option to do `z_init_static()` earlier pre `POST_KERNEL`?

**Impact**

Drivers with long initialization duration and lower significance are stretching out boot time. There is no way to push them to a later slot.

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

gmarull, 2025-02-20:
Consider using `zephyr,deferred-init` for the display. Also, adding a "lazy init" option to the device could be a solution. It sounds to me a driver-level problem, not something that needs to be fixed at a system-level.

Also, can't `z_init_static` just be moved before POST_KERNEL step is called? Is it done this way, so constructors can operate with devices (e.g. check ready)?

th-fischer2020, 2025-02-26:
Thank you for your advice. Deferred-init is what i was searching for.

closed 2025-02-26 by th-fischer2020