zephyrproject-rtos/zephyr #86043: k_thread_abort tests fail when split

fsammoura, 2025-02-20:
**Notes**
The k_thread_abort tests fail when split. This is the PR: https://github.com/zephyrproject-rtos/zephyr/pull/85803/commits/0ac1eb424b4cbdfb725d5b948db7ff0763b56e6f

**Describe the bug**
The k_thread_abort tests fail when split. This is the PR: https://github.com/zephyrproject-rtos/zephyr/pull/85803/commits/0ac1eb424b4cbdfb725d5b948db7ff0763b56e6f
It has to do with spinlock

**To Reproduce**
run tests on the file modified with this PR: https://github.com/zephyrproject-rtos/zephyr/pull/85803/commits/0ac1eb424b4cbdfb725d5b948db7ff0763b56e6f

**Expected behavior**
The tests should pass

**Impact**
cannot tests for k_thread_abort. Potential issues with k_thread_abort in real life usage.

**Logs and console output**
Check logs from PR tests

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

andyross, 2025-02-21:
Partial fix up at https://github.com/zephyrproject-rtos/zephyr/pull/86164

There are some arch layer fixes needed for riscv32/64, x86/x86_64 and sparc that prevent the test from passing there, so it's got a skip added.  But it works everywhere else.

andyross, 2025-02-21:
We can leave this open for a while.  Whether the arch fixing is really required or not is sort of an open question.  I'll bang on it a bit, time-available.  But IMHO we could close this without shame unless there's a desperate desire on the part of those architectures to run a self-abort-essential-thread test.

closed 2025-05-12 by github-actions[bot]