zephyrproject-rtos/zephyr #86051: tests\bluetooth\shell build error on mimxrt1170_evk@B/mimxrt1176/cm7 due to no default controller selected

JingsaiLu, 2025-02-20:
**Describe the bug**

tests\bluetooth\shell build error on mimxrt1170_evk@B/mimxrt1176/cm7 due to no default controller selected

**To Reproduce**

- west build -p always -b mimxrt1170_evk@B/mimxrt1176/cm7 tests\bluetooth\shell
- will see build error on the console. 

**Expected behavior**
- expect no build error for bluetooth shell west build on RT1170 as samples/bluetooth/handsfree
- may need to add a default controller in boards/.conf for the board

**Logs and console output**

[image]

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

dleach02, 2025-02-20:
@JingsaiLu, This isn't a bug. The RT1170 platform doesn't have a default. If you are expecting this to work I suspect you are attaching a BLE module at which point you need to create an overlay

closed 2025-02-20 by dleach02

JingsaiLu, 2025-02-20:
Thanks @dleach02 , 
but for the samples/bluetooth/handsfree, we have one default configuration for rt1170.
https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/handsfree/boards
which rule do we need to follow ? 
I think that one standard setup as a default may be more easy for newbies.

and is there any doc guide for this usage, other than from one error message ?