zephyrproject-rtos/zephyr #86052: RFC: drivers: stepper: API Specification Improvements

jbehrensnx, 2025-02-20:
## Introduction

As discussed in [PR 85658](https://github.com/zephyrproject-rtos/zephyr/pull/85658), here is the RFC about ambiguities concerning the expected behavior in the stepper api.

### Problem description

While the the stepper api has made significant improvements concerning its specification, some questions remain. The purpose of this RFC is to discuss these and find answers. This will then be used to improve the documentation, and refine the tests from [PR 85658](https://github.com/zephyrproject-rtos/zephyr/pull/85658), hopefully ensuring consistent behavior from all drivers that use the api.

### Proposed changes

Things that I myself have noticed:
- _set_microstep_interval_ will cause some drivers to immediately change their their speed if they are already moving, while the current documentation implies that the new speed only takes effect once a movement command (_move_to/move_by/run_) has been called. I prefer this second interpretation and would like the documentation to be clarified in this regard.
- _set_microstep_interval 0_ will fail with some drivers, which is in contrast to the old _set_speed 0_ which caused the motor to stop. As a result, currently the only way to stop a moving motor is to call disable followed by enable, which is not great. As a result, I would propose to either return _set_microstep_interval 0_ to its status as a stop command, or, preferably, to define a new dedicated _stop_ command that stops the motor and keeps the coils energized.
- regardless of the results of the previous issue, I would propose an _emergency_stop_ command that immediately stops the motor and keeps the coils energized. The difference to the previous entry is mainly with acceleration based drivers, such as tmc 50XX and the step-dir acceleration driver I am currently developing. The regular stop command would result in them gradually decelerating, but for safety reasons it might be necessary to halt them immediately.
- the _enable/disable_ command is completely unspecified in its behavior. My recommended behavior would be (based on the drv8424 driver):
    - _disable (enable off)_ causes movement to stop (including is_moving and changing position value), the motor coils to be de-energized and  movement commands (move_to/move_by/run) to fail with an error code (drv8424 currently uses -_ECANCELED_). Setting microstep interval and resolution still works however.
    - _enable_ causes the motor coils to be energized, but no movement occurs (even if the motor was moving before the disable command) until a movement command is called.

### Dependencies

Changes to the api will require modification of the existing drivers. Their limited number means that this will be manageable.

## Alternatives

Not improving the specification will lead to (and to a degree already has) divergent behavior of stepper api drivers, which is a significant problem.

jbehrensnx, 2025-02-20:
@jilaypandya @faxe1008 @Irockasingranite @fabiobaltieri @bjarki-andreasen

jilaypandya, 2025-02-20:
@jbehrensnx Thanks for creating this RFC.

I'll try to address the changes sequentially :)

---

> set_microstep_interval will cause some drivers to immediately change their their speed if they are already moving

True.

> while the current documentation implies that the new speed only takes effect once a movement command (move_to/move_by/run) has been called. I prefer this second interpretation and would like the documentation to be clarified in this regard.

Not really sure if this is the case in the API documentation as of now.

If a user wants to change micro_step_interval while the stepper is moving, i think it's completely okay.

---

> set_microstep_interval 0 will fail with some drivers which is in contrast to the old set_speed 0 which caused the motor to stop. As a result, currently the only way to stop a moving motor is to call [...]

> regardless of the results of the previous issue, I would propose an emergency_stop command that immediately stops the motor and keeps the coils energized. The difference to the previous entry is [...]

We can introduce a new dedicated `stop`(emergency_stop) command and keep the coils energized.

---

> the enable/disable command is completely unspecified in its behavior. My recommended behavior would be (based on the drv8424 driver):
> disable (enable off) causes movement to stop (including [...]

I think this is already the case as of now, if you try to call any of the move commands with motor disabled, -ECANCELED will be the return value as per stepper api.

> enable causes the motor coils to be energized, but no movement occurs (even if the motor was moving before the disable command) until a movement command is called.

This sounds fair :)

jbehrensnx, 2025-02-21:
>>set_microstep_interval will cause some drivers to immediately change their their speed if they are already moving

>True.

>>while the current documentation implies that the new speed only takes effect once a movement command (move_to/move_by/run) has been called. I prefer this second interpretation and would like the documentation to be clarified in this regard.

>Not really sure if this is the case in the API documentation as of now.
>If a user wants to change micro_step_interval while the stepper is moving, i think it's completely okay.

Just to clarify: your opinion is, that `set_microstep_interval` causing speed changes is the intended behavior? If yes than I would like it to be made explicit in the specification. If it should instead be implementation dependent, I would also like this to be made explicit and combined with adding an api call to determine which version is used by a given driver.
The point where this becomes relevant is, from my point of view, when you `run` the motor at a certain speed and then, while it is moving, want to `move_by/move_to` at a different speed. Not sure if it would be considered to be to contrived/niche.

>>the enable/disable command is completely unspecified in its behavior. My recommended behavior would be (based on the drv8424 driver):
disable (enable off) causes movement to stop (including is_moving and changing position value), the motor coils to be de-energized and movement commands (move_to/move_by/run) to fail with an error code (drv8424 currently uses -ECANCELED). Setting microstep interval and resolution still works however.

>I think this is already the case as of now, if you try to call any of the move commands with motor disabled, -ECANCELED will be the return value as per stepper api.

I mainly mentioned it because the the two tmc drivers don't do this, but yes the specification mentions the -ECANCELED. I should probably given this as an example of not following the api not being caught without tests in the test pr instead.

>- [ ]  Introduce a stop function in api

I would like to add both stop and emergency_stop as they have two different purposes, one to simply stop the motor during regular movement, and second to immediately stop the motor, regardless of the consequences. While these two functions would be the same for constant velocity drivers (gpio, counter + work timing source for step-dir), they would differ for acceleration based drivers like tmc50xx and the acceleration-counter timing source I am currently working on.

>- [ ] Update `enable `function with the specification : enable causes the motor coils to be energized, but no movement occurs (even if the motor was moving before the disable command) until a movement command is called.

I would like to also clarify what `enable off` does in the specification as well. While one could argue, that it is sort of specified in the movement commands, specifying it explicitly for the `enable` would make it clearer I think.

jilaypandya, 2025-02-21:
> Just to clarify: your opinion is, that `set_microstep_interval` causing speed changes is the intended behavior? If yes than I would like it to be made explicit in the specification. If it should [...]

Set the time interval between steps in microseconds(correction to be introduced: nanoseconds). Setting a time interval between two consecutive steps affects the speed of the stepper is evident imo.

> The point where this becomes relevant is, from my point of view, when you `run` the motor at a certain speed and then, while it is moving, want to `move_by/move_to` at a different speed. Not sure if it would be considered to be to contrived/niche.

so `run(dev,direction)` and `move(dev,usteps)`, as the name suggests only execute run or move commands, and the concern of setting speed is with `set_microstep_interval(dev, nsecs)`. 
Just from the top of my head, wouldn't a ramp driver want to periodically update interval(speed) while ramping up/down. ?

--- 

> I mainly mentioned it because the the two tmc drivers don't do this, but yes the specification mentions the -ECANCELED. I should probably given this as an example of not following the api not being caught without tests in the test pr instead.

Adding a todo for this

---

> I would like to add both stop and emergency_stop as they have two different purposes, one to simply stop the motor during regular movement, and second to immediately stop the motor, regardless of [...]

Took some inspiration from trinamic datasheet :)
```
enum stepper_stop_mode {
        SOFT_STOP,
        HARD_STOP,
}
stepper_stop(dev, stop_mode)
```

--- 

> I would like to also clarify what `enable off` does in the specification as well. While one could argue, that it is sort of specified in the movement commands, specifying it explicitly for the `enable` would make it clearer I think.

👍

jbehrensnx, 2025-02-24:
>Set the time interval between steps in microseconds(correction to be introduced: nanoseconds). Setting a time interval between two consecutive steps affects the speed of the stepper is evident imo.

I disagree that this evident, which is why I would prefer a mention that if the motor is moving, that the new speed takes immediate effect.

>so `run(dev,direction)` and `move(dev,usteps)`, as the name suggests only execute run or move commands, and the concern of setting speed is with `set_microstep_interval(dev, nsecs)`.
Just from the top of my head, wouldn't a ramp driver want to periodically update interval(speed) while ramping up/down. ?

The ramp driver would obviously need to know its current speed. Setting a new speed via `set_microstep_interval(dev, nsecs)` would cause the ramp driver to accelerate or decelerate to match this new speed. This means that current and target speed might differ, which is why in my current implementation for the step-dir acceleration driver the timing source can return the current interval (not yet implemented for the step-dir component, but that would be trivial) for internal calculations. For controlability reasons I would prefer to change the physical speed of the motor only once a new movement command is given. That said, the speed change taking immediate effect would be possible.

jilaypandya, 2025-02-24:
> I disagree that this evident, which is why I would prefer a mention that if the motor is moving, that the new speed takes immediate effect.

Okay, i'd like to briefly get the opinion of @faxe1008 on this. (since this function was introduced by him :) )

jbehrensnx, 2025-02-24:
>iirc the ramp controller we currently have in-tree the tmc50xx does not implement the set_step_interval function instead you are supposed to use the driver specific interface method. 

Internally we found the fact that tmc50xx does not support `set_step_interval` quite strange, though it should be mentioned that the acceleration driver that I am currently developing is quite a bit simpler, having only the acceleration as a configuration value, and that one rarely changes, so reusing `set_step_interval` makes a lot of sense in that case, as it simply sets the new target speed to accelerate towards.

>As for the step-dir driver: Sure adding that to the timing source "sub-api" does not hurt imo, but i am wondering if there is any precision to be gained from that. To know the "real" speed its not just about querying the current timing source settings but to know if the timing source has already elapsed "once" with the new settings, if that makes sense.

The step-dir-acceleration implementation actually uses that value for its calculation. I actually meant adding an api-call for getting the current interval, which might be useful even without acceleration, as the user wouldn't have to keep track of that value. There are also other things to consider, like triggering an event when the target speed is reached, but that would be part of the PR/RFC for that driver.
The "current speed" is always just a snapshot that can constantly change. The value is only valid for the moment the function call is made. That is just the limitation of a moving system. If you meant the values part in the current interval, the value always describes the the current interval between step signals, except the short time between triggering the interrupt and calculating the new value. As that runs in an interrupt context however, the get_interval function wouldn't get executed during that time.

jilaypandya, 2025-02-28:
Building upon https://github.com/zephyrproject-rtos/zephyr/pull/86431#issuecomment-2691139104 and https://github.com/zephyrproject-rtos/zephyr/pull/86495#issuecomment-2691184252

I would like to propose the following

1. Introduce a simple `stop`(coils stay energized) and `disable`(coils get energized) functions.
2. deprecate `is_moving function`
3. introduce get_stepper_state function as follows.
```
enum stepper_state {
    MOVING,
    NOT_MOVING,
    ENABLED,
    DISABLED,
    STOPPED,
}

int get_stepper_state(const struct device *dev, enum stepper_state *state);
```

jbehrensnx, 2025-03-05:
> Building upon [#86431 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/86431#issuecomment-2691139104) and [#86495 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/86495#issuecomment-2691184252)
> 
> [@dipakgmx](https://github.com/dipakgmx) [@bjarki-andreasen](https://github.com/bjarki-andreasen) [@fabiobaltieri](https://github.com/fabiobaltieri) [@jbehrensnx](https://github.com/jbehrensnx) [@faxe1008](https://github.com/faxe1008) [@kartben](https://github.com/kartben)
> 
> I would like to propose the following
> 
>     1. Introduce a simple `stop`(cancel movements, however coils stay energized) and `disable`(cancel movements and coils get deenergized) functions.
> 
>     2. deprecate `is_moving function`
> 
>     3. introduce get_stepper_state function as follows.
> 
> 
> ```
> enum stepper_state {
>     STANDSTILL,
>     MOVING,
>     ENABLED,
>     DISABLED,
>     STOPPED,
> }
> 
> int get_stepper_state(const struct device *dev, enum stepper_state *state);
> ```

I like the general idea, but I am not sure about the exact states you propose. `ENABLED` and `DISABLED` are clear. However, while `MOVING` the motor is still technically in the state `ENABLED`. Then there is the question of what the state `STOPPED` even means, the event in [PR 86431](https://github.com/zephyrproject-rtos/zephyr/pull/86431) should be enough for that. I am also unsure what the state `STANDSTILL` exactly means.
Maybe a bit array might be an idea? So you can set separate flags for enabled and `MOVING`. But I am not sure how good of an idea that is.
Properly documenting the enum is another option, but even then,the fact that the motor is technically in the state `ENABLED` and `MOVING` at the same time doesn't sit quite right with me,but I am not sure how to best resolve this.

Schulle73, 2025-03-05:
From the user's perspective. if you have Motorized Stages (like 8MT200-100) it's easy to define stepper_states. 

DISABLED -> off. In case of TMC5041, the Signal DRV_ENN is low. 
ENABLED -> stepper Driver is active TMC is in hold state (definition of IHOLD is necessary for this) 
MOVING -> Motor rotates, in cas of TMC move to defined position (definition of IRUN is necessary for this) 
STANDSTILL -> Motor can not drive, in case of TMC STALL detect

STOPPED is the same as ENABLED.
I'm wondering that you don't use the Interrupt Signal from the stepper. Interrupt outputs always have a reason. No one would come up with the idea of ​​polling an A/D converter to find out whether the conversion is finished.
polling is out. for years. 

I think stall detect interrupt is important. Stall detect is important for Motorized Stages to detect the end positions when neither a potentiometer nor a limit switch is used. Stall detect is always at the ends. You drive to both ends and then know how long the path is (in steps).

jilaypandya, 2025-03-05:
> From the user's perspective. if you have Motorized Stages (like 8MT200-100) it's easy to define stepper_states.
>
> DISABLED -> off. In case of TMC5041, the Signal DRV_ENN is low. ENABLED -> stepper [...]

You are right, bringing stall detect interrupt is in my personal backlog, i will try to soon bring it in. Would like to have end-detect support for gpio-controller as well through end-stops :)

Schulle73, 2025-03-06:
"end-detect support for gpio-controller" good idea. But make it flexible to have the two end positions and maybe more e.g. for the middle position.

jbehrensnx, 2025-03-06:
Personally I am iffy on having end stops in the API, as I feel that would be the task of a motion subsystem one level above. There is also the fact to consider, that many stepper drivers don't have end-points. I'd argue, that end point detection for now should be application code until a more general motor api or motion subsystem exists.
Stall detection is a different topic of course. The event already exists, drivers currently just don't trigger it.

Schulle73, 2025-03-06:
Correct, stepper don’t have endpoints. But the stepper is not a standalone device. The stepper is integrated into a system. But I don’t think that many stepper driven mechanics don't have end-points.
Steppers are mostly used for exact positioning in an range from x to y. The Endpoint  (x & Y) can be detect with an hardware switch. But the solution with the lowest cost don’t have these switches. A reference drive will be performed on power on to detect the endpoints. The stall detect feature is used for this.

jbehrensnx, 2025-03-06:
But the api is only about the stepper drivers, so drv8424, tmc2209 and so on. And in most cases they don't have end point support. Thus the stepper api shouldn't have them. What you are talking about is definitely something that a motion subsystem should handle, but for the stepper api it would be feature creep.
The task of the stepper api is to drive stepper motors using controllers like the ones named. Moving a certain distance in a direction and often detecting motor stalls. But higher level logic like detecting and managing end points is a level above this. This is one of the tasks of a motion subsystem - something that Zephyr is admittedly currently lacking.

Schulle73, 2025-03-07:
drv8424 and tmc2209 are very different. The drv8424 has a STEP/DIR interface only, the normal GPIO API should be enough.
The API's approach that the motor should travel a certain distance in one direction is definitely correct. If motor standstills are then detected, this also covers the detection of end points. If you don’t like the wording "end point support "then just say "motor stalls"
But when the motor stops (stalls), the issue of interrupts comes up again. nFAULT (DRV8425/5) is the magic signal.

jbehrensnx, 2025-03-07:
I am aware of the fault pin on the drv8424/25. That triggering a motor stalled event is indeed currently missing. What to do with the motor stalled event, besides calling the registered event callback, is however not the task of the stepper api. That would be the task of an api/subsystem that is one level higher (like a motion subsystem). Something like this currently does not exist and thus it would be the task of the application.
Thus I am of the opinion, that the api is fine in that regard, as it already supports the drivers sending stall events. The issue is instead that the drivers currently don;t implement stall detection.

Schulle73, 2025-03-07:
The stall detection should be implemented in the driver, that's clear. It can be useful to separate the stepper motor drivers and the stepper motor controllers. The stepper motor driver can only stop the motor by deactivating the H-bridge, which results in the motor coils being de-energized and the motor rotation being mechanically braked. 
And the movement between switching off and stopping is exactly what you don't want if you want precise positioning. Unless you spend money on position encoders:-)
The API for controllers can then be more detailed than for drivers if you separate them.

jbehrensnx, 2025-03-12:
Not sure if this is the correct RFC for this, but the `property-allowlist` should probably be removed from the step-dir driver bindings. As more timing sources, step-dir implementations and drivers are added, keeping this up to date for every driver will become a maintenance nightmare.

jilaypandya, 2025-03-13:
> Not sure if this is the correct RFC for this, but the `property-allowlist` should probably be removed from the step-dir driver bindings. As more timing sources, step-dir implementations and drivers are added, keeping this up to date for every driver will become a maintenance nightmare.

Not really sure what you mean :). all step-dir drivers need step-pin, dir-pin and hence they have been put in a common stepper-controller.yaml and you can get it via property-allowlist. Is there a reason why should each driver create this binding for self?

jbehrensnx, 2025-03-13:
> Not really sure what you mean :). all step-dir drivers need step-pin, dir-pin and hence they have been put in a common stepper-controller.yaml and you can get it via property-allowlist. Is there a [...]

I am not talking about doing away with importing properties with stepper-controller.yaml. The property-allowlist is a whitelist however. If a new timing source or step-dir implementation adds new properties, these will manually have to be added to every driver yamls property-allowlist , which is bad for maintenance. If stepper-controller.yaml is instead imported without property-allowlist, all properties are automatically available, requiring no further changes to the driver yaml files for new properties.

jilaypandya, 2025-03-16:
Hi @jbehrensnx, If all the concerns are addressed, i would like to close the issue :)

jbehrensnx, 2025-03-17:
All are addressed for now.

closed 2025-03-17 by jilaypandya