zephyrproject-rtos/zephyr #86059: mDNS doesn't work with iOS

VineetaNarkhede-eaton, 2025-02-20:
**Describe the bug**
When a query is sent from iOS using a hostname, no response is received. However, it works with an IP address. After debugging, I discovered that the query type (65-HTTPS) sent by iOS is not supported in the Zephyr code, causing the query to be rejected.

I could resolve this issue by adding support for the query type 65.

henrikbrixandersen, 2025-02-20:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

jukkar, 2025-02-20:
This is more like an enhancement than a bug so changing label.

PR #87268 (2025-03-20): net: mdns: Add support for iOS.
Fixes: #86059

The query type used by iOS is HTTPS-65,
Adding HTTPS query type to support iOS.
closed 2025-03-20 by carlescufi