zephyrproject-rtos/zephyr #86091: drivers: bluetooth: hci: ipc: HCI IPC driver drops incoming ACL data

PavelVPV, 2025-02-20:
**Describe the bug**
When building a bluetooth application for nRF5340 when Host and Controller are located application and network cores accordingly the HCI IPC driver is used to exchange HCI packets between network and application cores.

The HCI IPC driver is currently implemented in a way that it always expects a free buffer for the incoming (from Controller to Host) ACL data:
https://github.com/zephyrproject-rtos/zephyr/blob/059e86f426b8285e0c336c459baaff18dc3452e7/drivers/bluetooth/hci/ipc.c#L129-L137

I've implemented a test that sends notifications to a peer device over multiple EATT channels and makes the HCI IPC driver eventually drop the incoming ACL data. I put a PR with the test here: https://github.com/zephyrproject-rtos/zephyr/pull/85991

The test fails with and without `CONFIG_BT_HCI_ACL_FLOW_CONTROL` (see logs below).

The case with enabled `CONFIG_BT_HCI_ACL_FLOW_CONTROL` should be fixed by #83774 (if it gets merged).
The test fails with disabled `CONFIG_BT_HCI_ACL_FLOW_CONTROL` regardless of 83774 PR.

**To Reproduce**
To reproduce the dropped ACL data when `CONFIG_BT_HCI_ACL_FLOW_CONTROL` is **en**abled (if #83774 is not merged):
1. Cherry-pick https://github.com/zephyrproject-rtos/zephyr/pull/85991/commits/2dd0b63261db39587f6e755ab742bfafee739a62 (the test commit from #85991)
2. Build `tests/bsim/bluetooth/host/gatt/notify_stress` for `nrf5340bsim/nrf5340/cpuapp` target.
3. Observe `No available ACL buffers!` messages

To reproduce the dropped ACL data with `CONFIG_BT_HCI_ACL_FLOW_CONTROL` is **dis**abled:
1. Cherry-pick https://github.com/zephyrproject-rtos/zephyr/pull/85991/commits/2dd0b63261db39587f6e755ab742bfafee739a62 (the test commit from #85991)
2. Cherry-pick https://github.com/zephyrproject-rtos/zephyr/pull/86087/commits/0af5beb1929dd41d04aa68b98370bfbd4b47a2c6
3. Build `tests/bsim/bluetooth/host/gatt/notify_stress` for `nrf5340bsim/nrf5340/cpuapp` target.
4. Observe `No available ACL buffers!` messages

**Expected behavior**
The ACL data must never be dropped by an HCI driver.

**Impact**
A Bluetooth LE nRF5340 device using ACL data.

**Logs and console output**
The output from the test when `CONFIG_BT_HCI_ACL_FLOW_CONTROL` is **en**abled (run here: https://github.com/zephyrproject-rtos/zephyr/actions/runs/13421908610/job/37496285461?pr=85991):
```
...
d_00: @00:00:01.314845  (CPU:0): [00:00:01.314,819] <err> bt_hci_driver: No available ACL buffers!
d_00: @00:00:02.312836  (CPU:0): [00:00:02.312,835] <err> bt_conn: Unexpected first L2CAP frame
...
```

The output from the test when `CONFIG_BT_HCI_ACL_FLOW_CONTROL` is **dis**abled:
```
...
d_01: @00:00:01.309167  (CPU:0): Notifications enabled
d_01: @00:00:01.309388  (CPU:0): Sent notification #0 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #1 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #2 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #3 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #4 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #5 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #6 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #7 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #8 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #9 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #10 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #11 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #12 with length 60
d_01: @00:00:01.309388  (CPU:0): Sent notification #13 with length 60
d_00: @00:00:01.310073  (CPU:0): Subscribed to long characteristic
d_01: @00:00:01.310303  (CPU:0): Sent notification #14 with length 60
d_01: @00:00:01.311019  (CPU:0): Sent notification #15 with length 60
d_00: @00:00:01.311036  (CPU:0): Subscribed
d_00: @00:00:01.312817  (CPU:0): Received notification #0 with length 60
d_00: @00:00:01.324373  (CPU:0): [00:00:01.324,371] <err> bt_hci_driver: No available ACL buffers!
d_00: @00:00:01.324985  (CPU:0): [00:00:01.324,981] <err> bt_hci_driver: No available ACL buffers!
d_00: @00:00:01.325693  (CPU:0): [00:00:01.325,683] <err> bt_hci_driver: No available ACL buffers!
d_00: @00:00:01.326401  (CPU:0): [00:00:01.326,385] <err> bt_hci_driver: No available ACL buffers!
d_00: @00:00:01.327013  (CPU:0): [00:00:01.326,995] <err> bt_hci_driver: No available ACL buffers!
...
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.8
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/commit/059e86f426b8285e0c336c459baaff18dc3452e7

PavelVPV, 2025-02-20:
A proposed workaround to prevent HCI IPC driver drop packets: #86087

TorstenRobitzki, 2025-04-28:
I would love to see that bug being fixed.

PavelVPV, 2025-04-28:
> I would love to see that bug being fixed.

Hi @TorstenRobitzki , do you see it now even with `CONFIG_BT_HCI_ACL_FLOW_CONTROL` enabled?