zephyrproject-rtos/zephyr #86123: Sample `code_relocation_nocopy` doesn't work on STM32F746-DISCO

shashsuresh, 2025-02-21:
**Describe the bug**

When running the `code_relocation_nocopy` sample on the STM32F746 disco board, the application throws an instruction access violation, when trying to execute code from the QSPI flash.

**To Reproduce**

Steps to reproduce the behavior:
1. Create a `stm32f746g_disco.conf` in the `samples/application_development/code_relocation_nocopy/boards` directory.
2. Add the following configs to `stm32f746g_disco.conf` 
```
CONFIG_FLASH=y
CONFIG_STM32_MEMMAP=y
```
3. Build the sample
4. Make a serial connection with the STM32F746G-DISCO board (using screen/minicom/any other serial comm. tool)
5. Flash the sample

**Expected behavior**

**Impact**

**Logs and console output**

[image]

**Environment (please complete the following information):**

 - Fedora 40
 - Zephyr SDK 0.16.1
 - zephyrproject-rtos Commit: b1fe93e0470bde93926c321882a4c02ee33d2303

**Additional context**

rruuaanng, 2025-02-21:
@FRASTM

FRASTM, 2025-02-21:
The External Flash Memory @90000000  where the code is executed has to be declared with `ATTR_MPU_IO` attributes

FRASTM, 2025-02-21:
Are you sure the code is correctly downloaded ?

shashsuresh, 2025-02-25:
Programming with OpenOCD did the trick. (` west flash -r openocd`)

I used the default runner (which is the stm32cubeprogrammer I believe) the first time and looks like I did not set it up correctly flash to the external flash.

Thanks a lot for looking into and rectifying this issue so quickly.

Should I go ahead and close this issue? Or will it be closed once the PR #86136 is closed?

erwango, 2025-02-25:
> Should I go ahead and close this issue? Or will it be closed once the PR https://github.com/zephyrproject-rtos/zephyr/pull/86136 is closed?

Please wait #86136 is merged

closed 2025-03-18 by kartben