zephyrproject-rtos/zephyr #86167: Build fail when passing CONFIG_ string assignment from the command line

fabiobaltieri, 2025-02-21:
**Describe the bug**
Hey folks, looks like passing string configs from the command line broke with https://github.com/zephyrproject-rtos/zephyr/pull/85994

```console
$ west build -p -b nrf52dk/nrf52832 samples/hello_world/ -DCONFIG_BOOT_BANNER_STRING=\"broken\"
...
CMake Error at /home/fabiobaltieri/zephyrproject/zephyr/cmake/modules/yaml.cmake:552 (string):
  string sub-command JSON failed parsing json string: * Line 81, Column 134

    Missing ',' or '}' in object declaration

  .
```

added a print for `${json}` to figure what's going on and this is the problematic line:

```
    "command" : "/home/fabiobaltieri/.venv/bin/west build -p -b nrf52dk/nrf52832 samples/hello_world/ -DCONFIG_BOOT_BANNER_STRING=\\"broken\\"",
```

usual multiple escape issue it seems.

@tejlmand @pillo79 @nordicjm could any of you folks look into it?

Thanks!

**To Reproduce**
See above

**Expected behavior**
Build happily

**Impact**
It's pretty bad.

**Environment (please complete the following information):**
v4.1.0-rc2, Zephyr SDK, Linux, the usual

fabiobaltieri, 2025-02-21:
cc @kartben

kartben, 2025-02-22:
#86151 reports a similar issue

noahp, 2025-02-23:
Also impacting our CI, we've reverted for now.

gmarull, 2025-02-24:
related? https://github.com/zephyrproject-rtos/zephyr/issues/69552

tejlmand, 2025-02-24:
> related? [#69552](https://github.com/zephyrproject-rtos/zephyr/issues/69552)

I don't think so.

tejlmand, 2025-02-24:
looking at a fix.

fabiobaltieri, 2025-02-24:
m-torhan@ @noahp this should be fixed now, feel free to give it another go, thanks for reporting the issue.

PR #86243 (2025-02-24): cmake: do not double escape double quote
Fixes: #86167 

Do not escape single escaped quote `\"`.
A single escape quote in CMake indicates the use of literal `"`. Escaping those results in `\\"` which is a literal `\` and a quote which encapsulates a string.
This is a result we do not want.
Therefore, exempt the `\"` sequence from further escaping.