zephyrproject-rtos/zephyr #86172: MCUboot: configuration file ignored

JordanYates, 2025-02-22:
**Describe the bug**

When building an application with MCUboot support, the configuration file specified by MCUboot is overridden, resulting in build system warnings.

**To Reproduce**

```
west build -b nrf52840dk/nrf52840 samples/hello_world/ --sysbuild -p -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y
...
warning: MBEDTLS_CFG_FILE (defined at
/home/jordan/code/zephyr/bootloader/mcuboot/boot/zephyr/Kconfig:288, modules/mbedtls/Kconfig:52,
modules/mbedtls/Kconfig:52) was assigned the value 'mcuboot-mbedtls-cfg.h' but got the value
'config-tls-generic.h'. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MBEDTLS_CFG_FILE and/or look up
MBEDTLS_CFG_FILE in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.
...
```

**Expected behavior**

No build system warnings when building a default MCUboot application

**Environment (please complete the following information):**

 - Zephyr v4.1.0-rc2

de-nordic, 2025-02-25:
FYI @valeriosetti

valeriosetti, 2025-02-27:
It seems that the fix I proposed here https://github.com/zephyrproject-rtos/mcuboot/pull/124 also removes this warning. @de-nordic can you please take a look and let me know if that looks OK to you?
If so I can then open the proper PR as suggested by @nordicjm [here](https://github.com/zephyrproject-rtos/mcuboot/pull/124#issuecomment-2678327557)

closed 2025-05-20 by github-actions[bot]