zephyrproject-rtos/zephyr #86177: secure_storage: `secure_storage_its_uid_t` unneccesary abstraction

JordanYates, 2025-02-22:
**Is your enhancement proposal related to a problem? Please describe.**

The current definition of `secure_storage_its_uid_t` and the associated `secure_storage_its_caller_id_t` are attempting to solve a problem that doesn't appear to exist, and in doing so are creating their own problems.

https://github.com/zephyrproject-rtos/zephyr/blob/9b2fdace1d4c9686d43fbd8fad87b9620f576671/subsys/secure_storage/include/internal/zephyr/secure_storage/its/common.h#L25-L29
https://github.com/zephyrproject-rtos/zephyr/blob/9b2fdace1d4c9686d43fbd8fad87b9620f576671/subsys/secure_storage/include/internal/zephyr/secure_storage/its/common.h#L14-L23

The type of `psa_storage_uid_t` is defined as `uint64_t` by MbedTLS/TF-M, yet the implementation is worried about ID collisions in an address space of 18 quintillion options, and therefore adds an additional identifier on top. The original proposal in #76222 documents these choices under the **ID collisions** heading, but doesn't provide any examples of where collisions have been observed or are likely to occur.

If a subsystem/user requires up to 65536 unique keys (already a crazy amount), it can simply pick a 48 bit ID prefix at random and have a 0% chance of ever seeing a collision in practice (1 in 281 trillion).

The choice is a problem because the ability to actually store over 64 bits of information in an ID is not reasonable unless you are storing ID's as strings (i.e. the settings subsystem). The result of this is that ZMS, which can handle a crazy number of IDs due to its 64 bit ID space, is reduced to doing this for ID construction:
https://github.com/zephyrproject-rtos/zephyr/blob/9b2fdace1d4c9686d43fbd8fad87b9620f576671/subsys/secure_storage/src/its/store/zms.c#L55-L58

In the process of worrying about collisions on a 64bit address space, we are instead silently wrapping the ID's into a 32bit space, which in practice is simply stripping off the majority of the prefix that PSA ITS users expect to be preserved.

**Describe the solution you'd like**

Strip the `secure_storage_its_caller_id_t` caller ID entirely, implement `secure_storage` on a 64 bit ID.
Fix the ID wrapping in the ZMS backend.
Deal with any ID collisions by applying a random prefix to the user.

tomi-font, 2025-02-26:
> The type of psa_storage_uid_t is defined as uint64_t by MbedTLS/TF-M, yet the implementation is worried about ID collisions in an address space of 18 quintillion options, and therefore adds an [...]

That was done because:
- TF-M isolates different callers. So even if the PSA Crypto implementation calls the ITS with the same numerical ID than another caller (say, the application), the different callers are namespaced and no collision occurs in spite of the same ID. So one reason was for compatibility with TF-M.
- Also, without particular ID scheme, implementations might use the same IDs across PSA Crypto/ITS/PS APIs, resulting in collisions.

Now, the same could probably be achieved with the ID prefix approach you mention. When introducing the subsystem I didn't add any ID scheme at the same time but I still wanted to take care of that ID collision concern that had been brought up and so settled on that solution.

> The choice is a problem because the ability to actually store over 64 bits of information in an ID is not reasonable unless you are storing ID's as strings (i.e. the settings subsystem). The result [...]

Small correction: ZMS has 32-bit IDs, not 64.
And yes, I agree that the 64-bit ID space is absurdly large, and that it's not convenient to store even 64 bits as IDs.

What I did for the ZMS backend was in anticipation of https://github.com/zephyrproject-rtos/zephyr/pull/85581, which allows us to both limit the ID space to 32 bits (because it's already way enough) and, most importantly, to avoid all sorts of collisions by making the different users use allocated ID ranges.
It only applies to the PSA Crypto key IDs but could be extended to the ITS and PS.

> Even if all the above wasn't true, the addition of secure_storage_its_caller_id_t doesn't solve the problem of ID collisions, since applications built on TF-M can't make use of it, and therefore need to ensure ID uniqueness regardless.

This is true. `secure_storage_its_caller_id_t` wasn't solving the global issue of ID uniqueness. It was to make the secure storage implementation act like that of TF-M. https://github.com/zephyrproject-rtos/zephyr/pull/85581, on the other hand, solves the problem of ID collisions (for PSA Crypto keys) for TF-M as well.

> Strip the secure_storage_its_caller_id_t caller ID entirely, implement secure_storage on a 64 bit ID.
Fix the ID wrapping in the ZMS backend.
Deal with any ID collisions by applying a random prefix to the user.

I think we can do even better than the solution you propose. Whether getting rid of `secure_storage_its_caller_id_t` or not, we can make it so that the IDs that are handled by the secure storage subsystem are officially 32 bits (guarded by the fact that users of the APIs are expected to use their defined ID ranges).
Removing the ID wrapping from the ZMS backend, either moving it one level higher or getting rid of it completely (depending on how the IDs are allocated between the PSA Crypto/ITS/PS APIs).

Laczen, 2025-02-26:
The PSA key id's are defined as 32bit, the complete range has been reserved or is reserved for future use.

To use secure storage to store PSA keys and other items it is required that the secure storage id's are larger than 32 bit, that is why they are 64 bit. It is impossible to use zms id's directly for secure storage as they are only 32 bit.

It is possible to use zms for a PSA key storage, but it can't be used for secure storage.

closed 2025-08-11 by rlubos