zephyrproject-rtos/zephyr #86192: Zephyr v4.1.0 RC1 and RC2 don't boot on tested ESP32-S3 boards (working on v4.0.0)

mame82, 2025-02-22:
**Describe the bug**

Tested three ESP32-S3 boards which worked fine on zephyr v4.0.0 but none of the works on `v4.1.0-RC1` and `v4.1.0-RC2`.  I can't give any valuable input, as none of the boards manage to boot up to the `*** Booting Zephyr OS build ...` output, as they did on Zephyr `v4.0.0`.

To excclude errors with my app, I used `samples/basic/minimal` to verify the issue.

Boards tested:
- Waveshare ESP32-S3-touch-lcd-1-28
- Waveshare ESP32-S3-lcd-1-28 (custom device tree based on the touch version, basically only touch device removed and SPI pins for LCD and backlight adjusted to the board)
- Espressif ESP32-S3-DEvKitC

 

**To Reproduce**

- adjust `west.yaml` to use `revision: v4.1.0-rc1` or `revision: v4.1.0-rc2`
- run west update
- run `west blobs fetch hal_espressif` to update binary blobs accordingly
- build `samples/basic/minimal` for one of the aforementioned boards
- flash to the board (works with no issues up till here)
- run `west espressif monitor -p <correct serial port>` board boots bl, but does not progress to zephyr OS

- build samples/

**Expected behavior**

ESP32-S3 boards are able to boot Zephyr

**Impact**

3 out of 3 ESP32-S3 boards don't boot Zephyr v4.1.0-RC1/RC2 (all of them work on v4.0.0)

**Logs and console output**

Here's the example output of the serial port of `ESP32-S3-touch-lcd-1-28` after building and flashing (no further output after this). Board config `esp32s3_touch_lcd_1_28\esp32s3\procpu`:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fc8f470,len:0x1700
load:0x40374000,len:0xb460
SHA-256 comparison failed:
Calculated: dea8080824eebfe36d1499325bacc8b9d60dd6fea01773a16cbdb74fadd438dd
Expected: 0000000060340000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x403791ec
I (60) soc_init: ESP Simple boot
I (60) soc_init: compile time Feb 22 2025 22:19:11
W (60) soc_init: Unicore bootloader
I (60) soc_init: chip revision: v0.2
I (62) flash_init: Boot SPI Speed : 80MHz
I (66) flash_init: SPI Mode       : DIO
I (70) flash_init: SPI Flash Size : 16MB
I (73) boot: DRAM: lma 0x00000020 vma 0x3fc8f470 len 0x1700   (5888)
I (79) boot: IRAM: lma 0x00001728 vma 0x40374000 len 0xb460   (46176)
I (86) boot: IRAM: lma 0x0000cb98 vma 0x00000000 len 0x3460   (13408)
I (92) boot: IMAP: lma 0x00010000 vma 0x42000000 len 0x50d0   (20688)
I (98) boot: IRAM: lma 0x000150d8 vma 0x00000000 len 0xaf20   (44832)
I (104) boot: DMAP: lma 0x00020000 vma 0x3c010000 len 0x1070   (4208)
I (110) boot: Image with 6 segments
I (113) boot: IROM segment: paddr=00010000h, vaddr=42000000h, size=050CEh ( 20686) map
I (121) boot: DROM segment: paddr=00020000h, vaddr=3c010000h, size=01070h (  4208) map
I (140) boot: libc heap size 339 kB.
```

**Environment (please complete the following information):**

- Win11
- Zephyr v4.1.0-RC1 / v4.1.0-RC2
- OS: Windows 11

mame82, 2025-02-23:
Seems issue is not related to Zephyr directly. I am using "Zephyr IDE" extension for VScode. The build configuration lets you choose between "Debug", "Size" and "Speed" with "debug" set as the default (which worked on Zephyr v4.0.0). When "size" is used as build configuration, it works again.

zhang-wenchao, 2025-02-23:
I have this problem on esp32c3, same error. Stuck at boot: libc heap size xx kB. CONFIG_DEBUG=n can boot normally.

sylvioalves, 2025-02-24:
@mame82 Thanks for reporting. Submitted a fix linked to this issue. Please take a look if you can.

mame82, 2025-02-24:
Changed my west.yml to:

```
manifest:
  defaults:
    remote: zephyrproject
  remotes:
    - name: zephyrproject
      url-base: https://github.com/zephyrproject-rtos
  projects:
    - name: zephyr
      repo-path: zephyr
      remote: zephyrproject
      revision: pull/86202/head
      # revision: v4.1.0-rc2
      clone-depth: 1
      import:
        path-prefix: external
        name-allowlist:
          - cmsis
          - cmsis-dsp
          - picolibc
          - segger
          - hal_espressif
          - mbedtls
          - lvgl
```

... and ran `west update`.

Tested build with `-DCONFIG_DEBUG_OPTIMIZATIONS=y -DCONFIG_DEBUG_THREAD_INFO=y` and `--sysbuild -DCONFIG_DEBUG_OPTIMIZATIONS=y -DCONFIG_DEBUG_THREAD_INFO=y`.

Resulting image succesfully booted into Zephyr `*** Booting Zephyr OS build 0622edc9ab53 ***`.

Ran my test on "Waveshare ESP32-S3 Touch LCD 1.28". Unfortunately I have no none "ESP32-S3" based boards around, but at least for ES32-S3 issue is solved.

Thanks for your efforts.

PR #86202 (2025-02-26): soc: espressif: fix optimization flag boot fault
When DEBUG_OPTIMIZATION or NO_OPTIMIZATION is
enabled, efuse reading fails during bootloader start. Move those calls into IRAM area so that reading when cache is disabled works without any faults.

In HAL side, we need to use low level calls to read CPU id instead of Zephyr's default one.

Fixes #86192
closed 2025-02-26 by kartben