zephyrproject-rtos/zephyr #86210: esp32c3 does not build with mcuboot and CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256

tqjp, 2025-02-24:
**Describe the bug**
esp32c3 IRAM overflows when using mcuboot and CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256

**To Reproduce**
Adding
```
SB_CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256=y
```
to `sysbuild.conf`

**Expected behavior**
Build completes without errors.

**Logs and console output**
```
`.iram0.text' will not fit in region `iram_seg'
```

**Environment (please complete the following information):**

 - OS: Ubuntu
 - Zephyr 4.0.0

**Additional context**
Increasing `BOOTLOADER_IRAM_SEG_LEN`  in `zephyr/soc/espressif/esp32c3/memory.h` similarly as described in https://github.com/zephyrproject-rtos/zephyr/issues/76566 for the esp32s3 to `0xb000` solves the issue.
Why isn't there a CONFIG setting or similar to set those values?

zhang-wenchao, 2025-02-24:
https://github.com/zephyrproject-rtos/zephyr/pull/85288

tqjp, 2025-02-24:
> If this is the Zephyr port, the one built with `--sysbuild`, then secure boot it is not yet supported. This is the "software based" secureboot, and it is not on our roadmap. It could even be working, but we are not adding energy to test it due to Espressif port.

It does actually work, as you point out, but with the higher IRAM settings.
Do I understand this correctly, even checking images only once for FOTA is considered secure boot?
We are not doing a full stack secure boot with a signed mcuboot.

And what is meant with `basic cyber security`, see `MCUBOOT (Zephyr port, basic cyber security)` in https://github.com/zephyrproject-rtos/zephyr/issues/29394?

rftafas, 2025-02-24:
### Zephyr Port of MCUboot on ESP32 devices

MCUboot's Secure Boot (soft based) will prevent a bogus/hacked image to be loaded on the device. For example, the attacker tried to tamper, send a compromised image OTA to the device for update. The keys wouldn't match, no update will be made and device is safe (Cybersecurity). But one with a cable could go there and re-flash the device. Or remove the Flash chip and copy its contents.

Ok, the smart will remind me that there is e-fuse option for disabling flashing - but that is irreversible for ALL, attackers and non-attackers. And it is ineffective against flash removal and copying.

IMPORTANT:

- For non-Espressif devices, check their doc. This info is specific to Espressif devices, NOT to overall MCUboot.
- This secure boot implementation  is not supported. Could work, but we see no reason to do it because we have the Espressif port.

### Espressif Port of MCUboot

Espressif Ports uses hardware backed SecurebootV2, which offers physical and cyber security. You can't flash a bogus image, and don't need to burn e-fuses.

It also offers Flash Encryption, so only that specific ESP32 knows what is written to its flash - no attack by removing the flash chip. (This is on my favorite option: let the device randomly select its key and only it will know what it is written).

tqjp, 2025-02-24:
Thank you for the detailed description. We will have a look at the Espressif Port then.

Although it seems that for the esp32s3 the change in `memory.h` has been made for exactly this reason (https://github.com/zephyrproject-rtos/zephyr/issues/76566). Probably that is why @zhang-wenchao submitted a similar pull request for the esp32-c3. And in https://github.com/zephyrproject-rtos/zephyr/issues/29394 the Zephyr port / basic cybersecurity is marked as implemented and supported, I guess this all adds to the confusion.

zhang-wenchao, 2025-02-25:
I think this basic functionality is needed, although without secure boot, the existing implementation can provide basic protection for Bluetooth OTA after disabling debugging and disabling UART firmware downloads.

tqjp, 2025-02-25:
> I think this basic functionality is needed, although without secure boot, the existing implementation can provide basic protection for Bluetooth OTA after disabling debugging and disabling UART firmware downloads.

Try with zephyr v4.1.0-rc2. I just saw that the IRAM value has been increased a little bit since V4.0.0 and it seems to work, or at least it builds without errors, in our use case.

rftafas, 2025-02-25:
> I think this basic functionality is needed

It is already covered by Espressif Port.

Although we are pending the BT support on Espressif port, it is far more likely for us to add BT to Espressif Port than add any functionality in Zephyr Port.

closed 2025-05-11 by github-actions[bot]