zephyrproject-rtos/zephyr #86223: Clarify expected behavior when `CONFIG_WDT_DISABLE_AT_BOOT=n`

mathieuchopstm, 2025-02-24:
**Is your enhancement proposal related to a problem? Please describe.**
The current description for Kconfig symbol `CONFIG_WDT_DISABLE_AT_BOOT` is:
https://github.com/zephyrproject-rtos/zephyr/blob/9b2fdace1d4c9686d43fbd8fad87b9620f576671/drivers/watchdog/Kconfig#L17-L21

Expected behavior when `CONFIG_WDT_DISABLE_AT_BOOT=y` can be easily understood: WDT drivers __must disable__ the timer.
However, the expected behavior when `CONFIG_WDT_DISABLE_AT_BOOT=n` is not specified and could be either of:
* _Do not disable the WDT during boot_ [if it is enabled by default]
  * i.e., do nothing
* _Enable the WDT during boot_

The currently implemented behavior varies from driver to driver.

**Describe the solution you'd like**
The expected driver behavior when `CONFIG_WDT_DISABLE_AT_BOOT=n` should be agreed upon and documented.
Non-conforming drivers should be updated to implemented the agreed upon and documented behavior.

**Describe alternatives you've considered**
1. Keep expected behavior when `CONFIG_WDT_DISABLE_AT_BOOT=n` undocumented.

2. Replace `WDT_DISABLE_AT_BOOT` with more generic and explicit `choice WDT_BOOT_BEHAVIOR` such as:
* `WDT_HARDWARE_DEFAULT`
  * Driver should configure the WDT (e.g., reload counter, prescaler, etc)
  * Driver __must not__ disable or enable the WDT
  * After boot, the WDT status will depend on default hardware behavior
    * If WDT is enabled by default on SoC, it will be enabled
    * If WDT is disabled by default on SoC, it will be disabled
* `WDT_DISABLE_AT_BOOT`
  * Driver should not configure the WDT, and __must__ explicitly disable it
  * This option can be gated behind `HAS_WDT_DISABLE_AT_BOOT`
* `WDT_ENABLE_AT_BOOT`
  * Driver should configure and __must__ explicitly enable (if applicable) the WDT

Unanswered question: where should the default configuration come from? (STM32 has platform-specific Kconfig for this)

**Additional context**
The `samples/drivers/watchdog` and `tests/drivers/watchdog/wdt_basic_api` applications are setting `CONFIG_WDT_DISABLE_AT_BOOT=n`.
Since commit 2509de7d3e134f25eba4c9e132e13892f3898df5, these tests will "fail" on STM32 platforms, because the STM32 IWDG timer driver interprets `CONFIG_WDT_DISABLE_AT_BOOT=n` as _enable the WDT during boot_.
Since the timer can be configured only once, `wdt_install_timeout` returns `-ENOMEM` and a test failure is reported... despite the board getting reset by WDT after a little while. Replacing with `CONFIG_WDT_DISABLE_AT_BOOT=y` allows the tests to succeed.

In these two applications, `CONFIG_WDT_DISABLE_AT_BOOT=n` seems to have been set under the interpretation "_do not disable the WDT if enabled by default_", which breaks when the underlying implementation understands it as "_enable and configure the WDT_".

erwango, 2025-10-03:
@martinjaeger Would you have a recommendation on the way forward on this ?
Since sample is failing on STM32, the default interpretation is that STM32 behavior is faulty. Is that the way it should be understood and hence fixed ?

fberndl, 2025-10-08:
+1
This is currently not very intuitive , they behaviour of the samples seemed to change by upgrading from v4.0.0 to v4.2.0 and did cost me some time until I understood why CONFIG_WDT_DISABLE_AT_BOOT=y is necessary

yvesll, 2025-11-04:
For `wdt_mcux_cop.c`, here is a discussion: https://github.com/zephyrproject-rtos/zephyr/pull/90736#issuecomment-2916186931