zephyrproject-rtos/zephyr #86236: Getting Started Guide: Stuck on step 7 (west packages pip --install)

TK5090, 2025-02-24:
Hello,

I am installing every dependencies of Zephyr and Zephyr SDK on a docker Image.
Unfortunatelly, on step 7, I am getting an error.

The command on step 7 is:

`west packages pip --install`

I am getting the error code:
`Step 12/12 : RUN zephyrproject\.venv\Scripts\activate.bat ;     west packages pip --install
 ---> Running in fe1d767ff8f0
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-q] [-V] <command> ...
west: unknown command "packages"; do you need to run this inside a workspace?
The command 'powershell -Command zephyrproject\.venv\Scripts\activate.bat ;     west packages pip --install' returned a non-zero code: 1`

What is my mistake?
The Image uses windows as OS, not Linux.

kartben, 2025-02-25:
@pdgendt can you please help?

pdgendt, 2025-02-25:
Which version of Zephyr do you have? `west packages` has been introduced in `v4.1.0`. Before this you need to run
```shell
$ pip install -r zephyr/scripts/requirements.txt
```