zephyrproject-rtos/zephyr #86245: Add Wireguard VPN files to Zehyr

jukkar, 2025-02-24:
## Origin

wireguard-lwip project

https://github.com/smartalock/wireguard-lwip

## Purpose

Add support of Wireguard VPN to Zephyr.

## Mode of integration

Add and port relevant files from wireguard-lwip project to `subsys/net/lib/wireguard/` and its sub-directories in Zephyr main tree.
The Wireguard VPN is tightly integrated into Zephyr network stack so moving it as a module is not practical.

## Maintainership

Zephyr network maintainer @jukkar will maintain the code.

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/86020

## Description

The ported files from wireguard-lwip project contain functions that call the crypto code and the actual crypto code that implement the needed algorithms.

Files:
* `subsys/net/lib/wireguard/wg_crypto.c` contains functions that verify the integrity of the received packets, and decrypt and encrypt sent / received Wireguard network packets
* `subsys/net/lib/wireguard/crypto` directory and its sub-directories contain the implementation of actual needed crypto algorithms
* all the other files and sample code integrate the Wireguard VPN support to Zephyr, and are using Apache-2 license.

## Dependencies

None

## Revision

Version 0c44df308984abfb9c9df628002cdf8154c55506 (from 13 Nov 2022) of wireguard-lwip project is the head that was used in the porting.

## License

``BSD-3-Clause``
``MIT``

nashif, 2025-03-05:
This includes files that have public domain license! 
Per the guidelines in https://docs.zephyrproject.org/latest/contribute/external.html#software-license

_"The board will automatically reject licenses that have not been approved by the [Open Source Initiative (OSI)](https://opensource.org/licenses/alphabetical)."_

jukkar, 2025-03-06:
There are these files that have dual license, so user can either use Public Domain or MIT license. I think we can / should use the MIT here, the license check script does not seem to be able to detect something like this. All the other files have either BSD-3 or MIT license.

```
subsys/net/lib/wireguard/crypto/refc/poly1305-donna.c
1:/*  Taken from https://github.com/floodyberry/poly1305-donna - public domain or MIT */

subsys/net/lib/wireguard/crypto/refc/poly1305-donna-32.h
1:/* Taken from https://github.com/floodyberry/poly1305-donna - public domain or MIT */

subsys/net/lib/wireguard/crypto/refc/poly1305-donna.h
1:/*  Taken from https://github.com/floodyberry/poly1305-donna - public domain or MIT */
```

@nashif would this be acceptable?

nashif, 2025-03-06:
> subsys/net/lib/wireguard/crypto/refc/poly1305-donna-32.h
> 1:/* Taken from https://github.com/floodyberry/poly1305-donna - public domain or MIT */

this one needs SPDX, similar to the others.

> @nashif would this be acceptable?

yeah, I think so. We need some language in the external license docs explaining all of this https://docs.zephyrproject.org/latest/LICENSING.html#zephyr-licensing

nashif, 2025-03-06:
> ## License
> `BSD-3-Clause` `MIT`

need to add the Public domain part here, i.e. MIT or Public Domain.

jukkar, 2025-03-06:
> need to add the Public domain part here, i.e. MIT or Public Domain.

Done. I will add the missing SPDX line and resubmit the PR.

jukkar, 2025-03-06:
Updated the PR. Please take a look at https://github.com/zephyrproject-rtos/zephyr/pull/86020/commits/c2c24b2c77dc0d98fc00dee2d36cba2a569ee886 whether it looks ok.