zephyrproject-rtos/zephyr #86258: ESP32-C6: Enabling CONFIG_WATCHDOG=y breaks WiFi

ASerbinski, 2025-02-24:
**Describe the bug**

As the title indicates. Wifi works normally on ESP32-C6, unless setting CONFIG_WATCHDOG=y, which if set results in no wifi communications. Router log indicates no activity.

**To Reproduce**

Any wifi sample; set CONFIG_WATCHDOG=y

**Expected behavior**

Using the watchdog timer should not break wifi.

**Impact**

showstopper

**Logs and console output**

From samples/net/wifi/apsta_mode with the following patch for C6;
```
Subject: [PATCH] Add esp32c6 to apsta sample

---
 .../wifi/apsta_mode/boards/esp32c6_devkitc.conf | 17 +++++++++++++++++
 .../apsta_mode/boards/esp32c6_devkitc.overlay   | 10 ++++++++++
 2 files changed, 27 insertions(+)
 create mode 100644 samples/net/wifi/apsta_mode/boards/esp32c6_devkitc.conf
 create mode 100644 samples/net/wifi/apsta_mode/boards/esp32c6_devkitc.overlay

diff --git a/samples/net/wifi/apsta_mode/boards/esp32c6_devkitc.conf b/samples/net/wifi/apsta_mode/boards/esp32c6_devkitc.conf
new file mode 100644
index 00000000000..f1f8556ee47
--- /dev/null
+++ b/samples/net/wifi/apsta_mode/boards/esp32c6_devkitc.conf
@@ -0,0 +1,17 @@
+CONFIG_BOOTLOADER_MCUBOOT=y
+CONFIG_WIFI=y
+CONFIG_GPIO=y
+
+CONFIG_NETWORKING=y
+CONFIG_NET_L2_ETHERNET=y
+
+CONFIG_ESP32_WIFI_STA_AUTO_DHCPV4=y
+CONFIG_NET_MGMT_EVENT_QUEUE_SIZE=32
+CONFIG_HEAP_MEM_POOL_SIZE=98304
+
+CONFIG_WIFI_ESP32=y
+CONFIG_ESP32_WIFI_AP_STA_MODE=y
+CONFIG_WIFI_NM=y
+CONFIG_WIFI_NM_MAX_MANAGED_INTERFACES=2
+
+
diff --git a/samples/net/wifi/apsta_mode/boards/esp32c6_devkitc.overlay b/samples/net/wifi/apsta_mode/boards/esp32c6_devkitc.overlay
new file mode 100644
index 00000000000..a53651c69a2
--- /dev/null
+++ b/samples/net/wifi/apsta_mode/boards/esp32c6_devkitc.overlay
@@ -0,0 +1,10 @@
+&wifi {
+	status = "okay";
+};
+
+/ {
+	wifi_ap: wifi_ap {
+		compatible = "espressif,esp32-wifi";
+		status = "okay";
+	};
+};
-- 
2.48.1
```

Success (```#CONFIG_WATCHDOG is not set```):

```
*** Booting Zephyr OS build v4.1.0-rc2-19-gee8266797c17 ***
[00:00:05.039,000] <inf> MAIN: Turning on AP Mode
[00:00:05.039,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: Started DHCPv4 server, address pool:
[00:00:05.039,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: 	 0: 192.168.4.11
[00:00:05.039,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: 	 1: 192.168.4.12
[00:00:05.039,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: 	 2: 192.168.4.13
[00:00:05.039,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: 	 3: 192.168.4.14
[00:00:05.039,000] <inf> MAIN: DHCPv4 server started...

[00:00:05.089,000] <err> esp32_wifi: Failed to send packet
[00:00:05.090,000] <err> esp32_wifi: Failed to send packet
[00:00:05.090,000] <err> esp32_wifi: Failed to send packet
[00:00:05.090,000] <err> esp32_wifi: Failed to send packet
[00:00:05.090,000] <inf> MAIN: AP Mode is enabled. Waiting for station to connect
[00:00:05.090,000] <inf> MAIN: Connecting to SSID: ANET

[00:00:05.091,000] <err> esp32_wifi: Failed to send packet
[00:00:05.091,000] <err> esp32_wifi: Failed to send packet
[00:00:05.091,000] <err> esp32_wifi: Failed to send packet
[00:00:05.091,000] <err> esp32_wifi: Failed to send packet
[00:00:06.091,000] <err> esp32_wifi: Failed to send packet
[00:00:07.092,000] <err> esp32_wifi: Failed to send packet
[00:00:10.240,000] <inf> net_dhcpv4: Received: 192.168.1.84
[00:00:10.240,000] <inf> MAIN: Connected to ANET
```

Failure (```CONFIG_WATCHDOG=y```):
```
*** Booting Zephyr OS build v4.1.0-rc2-19-gee8266797c17 ***
[00:00:05.039,000] <inf> MAIN: Turning on AP Mode
[00:00:05.039,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: Started DHCPv4 server, address pool:
[00:00:05.039,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: 	 0: 192.168.4.11
[00:00:05.039,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: 	 1: 192.168.4.12
[00:00:05.039,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: 	 2: 192.168.4.13
[00:00:05.039,000] <dbg> net_dhcpv4_server: net_dhcpv4_server_start: 	 3: 192.168.4.14
[00:00:05.039,000] <inf> MAIN: DHCPv4 server started...

[00:00:05.091,000] <err> esp32_wifi: Failed to send packet
[00:00:05.091,000] <err> esp32_wifi: Failed to send packet
[00:00:05.091,000] <err> esp32_wifi: Failed to send packet
[00:00:05.092,000] <err> esp32_wifi: Failed to send packet
[00:00:05.092,000] <inf> MAIN: AP Mode is enabled. Waiting for station to connect
[00:00:05.092,000] <inf> MAIN: Connecting to SSID: ANET

[00:00:05.093,000] <err> esp32_wifi: Failed to send packet
[00:00:05.093,000] <err> esp32_wifi: Failed to send packet
[00:00:05.093,000] <err> esp32_wifi: Failed to send packet
[00:00:05.093,000] <err> esp32_wifi: Failed to send packet
[00:00:06.093,000] <err> esp32_wifi: Failed to send packet
[00:00:07.094,000] <err> esp32_wifi: Failed to send packet
```

fabiobaltieri, 2025-02-25:
Is this a regression? Any chance you could try and bisect it?

ASerbinski, 2025-02-25:
Its not a regression. Its never worked on the C6.

My project works on Xtensa and C3 versions of ESP32. While trying to build it for C6 immediately after wifi support was added to zephyr, I found that the wifi didn't work in my project. It was just yesterday that I had time to track it down to the watchdog.

phb98, 2025-02-26:
ap_sta coexist was broken last time i try. I found another config enable/disable ipv6 somehow mess with wifi too. Basically wifi is broken on C6 i believe.

ASerbinski, 2025-02-26:
No @phb98 

WiFi is working on C6, including APSTA. You can refer to the logs in the first post showing working output. Besides which, this bug report is about enabling watchdog.

sylvioalves, 2025-03-01:
@ASerbinski would you check this? #86517

ASerbinski, 2025-03-01:
Thank you @sylvioalves !!!

That fixes it!

PR #86517 (2025-03-06): soc: esp32c6: fix wifi by proper interrupt handling
Current interrupt allocator is not taking into account reserved areas.
In case of esp32c6, Wi-Fi isn't properly configured, causing instability or even non-functional feature.
This adds the reserved area ranges for all risc-v based SoC and unify the slot finding based on interrupt source.
This also fix the scenario where Wi-Fi is disconnected and esp_wifi_send() is called.

Fixes #86258
Fixes #85466
Fixes #83784
closed 2025-03-06 by fabiobaltieri