zephyrproject-rtos/zephyr #86265: Clean up some redundant descriptions in the bindings

rruuaanng, 2025-02-25:
**Is your enhancement proposal related to a problem? Please describe.**

There are still many descriptions like the one below in the current `dts/bindings`, which are actually too redundant. e.g.:
- `This is a representation of`
- `.... node`
- `This binding gives a representation of`

I think the best description would be this:
- `nRF7002 Wi-Fi chip with QSPI interface`
- `STM32 USB HS PHY controller`
- `ITE it8xxx2 PECI`

In other words, describe the hardware itself without excessive explanation.

**Describe the solution you'd like**

Let's clean them up! There are too many of them. Let's tidy up the list items while cleaning them up
— I'm just trying to take the easy way out.

- [x] cpu https://github.com/zephyrproject-rtos/zephyr/pull/86215

**Describe alternatives you've considered**

**Additional context**

henrikbrixandersen, 2025-02-25:
I would argue that we should also introduce a "title" tag for the short title of the binding while doing this, similar as to what is used in DTSchema, and reserve the "description" for a longer text, complete with binding examples and so on. Thoughts? @mbolivar

rruuaanng, 2025-02-27:
I noticed that some bindings distinguish between the title and description using this format, such as in `nxp,mbox-mailbox.yaml`:

```
description: |
  NXP Mailbox Unit as Zephyr MBOX.

  This NXP Mailbox driver implements Multi-Channel Inter-Processor Mailbox (MBOX) API
  around NXP Inter-CPU Mailbox peripheral IP block.

  The NXP Inter-CPU Mailbox provides up to thirty-two user defined interrupts.
  This driver uses 4 interrupts for mbox signalling mode per each channel,
  4 interrupts for mxbox data transfer mode per each channel and 24 as 3 bytes
  for data.
```
(Maybe it would be enough to just add some practical guidelines for bindings?)

decsny, 2025-02-27:
I personally think all of these can be done in one PR, and maybe even one commit, if its the same few words being deleted in every file

kartben, 2025-02-28:
> I would argue that we should also introduce a "title" tag for the short title of the binding while doing this, similar as to what is used in DTSchema, and reserve the "description" for a longer text, complete with binding examples and so on. Thoughts? @mbolivar 

+1 
But like @rruuaanng suggested I think a reasonable first step, which can only make the move to using a `title` property easier, is to generalize the use of the following format:

```
Short sentence (aka title).

More text.
After.
A blank line.
```

rruuaanng, 2025-03-02:
> +1 But like [@rruuaanng](https://github.com/rruuaanng) suggested I think a reasonable first step, which can only make the move to using a `title` property easier, is to generalize the use of the [...]

Oh, alright. It looks like everyone likes this. I'll get it done in my spare time, at least after most of my PRs are taken care of.
(I don't think this is particularly urgent.)

rruuaanng, 2025-03-02:
It seems that the `title` keyword was deprecated in the past for some reason. Do we still need to restore it? If the answer is yes, I think we should make title an optional keyword.
```
        # Descriptive errors for legacy bindings.
        legacy_errors = {
            "#cells": "expected *-cells syntax",
            "child": "use 'bus: <bus>' instead",
            "child-bus": "use 'bus: <bus>' instead",
            "parent": "use 'on-bus: <bus>' instead",
            "parent-bus": "use 'on-bus: <bus>' instead",
            "sub-node": "use 'child-binding' instead",
            "title": "use 'description' instead",
        }
```

mbolivar, 2025-03-27:
@rruuaanng ah, I see now that https://github.com/zephyrproject-rtos/zephyr/pull/86533 is related to this issue -- this would have been helpful for me to know when reviewing that PR.

I think the discussion in this issue should be enough for you to make the documentation updates I requested in the PR -- think you can make that work?

Probably also worth adding a `fixes: #86265` to #86533 's issue description in case other people are confused as I was.

PR #86533 (2025-04-02): edtlib: binding: Add a title keyword
Add a 'title' keyword to the binding to provide a short description of the binding, while 'description' serves as the long description.

fixes: #86265