zephyrproject-rtos/zephyr #86311: Disabling `GEN_ISR_TABLES` on Nordic Hardware

RobinKastberg, 2025-02-25:
**Describe the bug**
Currently, it seems like `CONFIG_GEN_ISR_TABLES=n` on `arm/cortex_m` should be a valid configuration.
But when trying to run this configuration using IAR (since we had linking problems with discarding .intList, this seemed like a reasonable workaround) we had some problems on Nordic hardware and abandoned the idea.

Is this a reasonable use case or does everyone on ARM just assume `GEN_ISR_TABLES` and I have misunderstood something?

**To Reproduce**

Zephyr toolchain:
```
 west build -p -b qemu_cortex_m0 tests/kernel/mutex/mutex_api -- -DCONFIG_GEN_ISR_TABLES=n
[...]
ninja run
user@b7f43aa2a597:/workdir/zephyr$ cd build/
user@b7f43aa2a597:/workdir/zephyr/build$ ninja run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m0
E: ***** HARD FAULT *****
E: r0/a1:  0x00000000  r1/a2:  0x20001d70  r2/a3:  0x20001d74
E: r3/a4:  0x00005a91 r12/ip:  0x00000000 r14/lr:  0xfffffffd
E:  xpsr:  0x60000018
E: Faulting instruction address (r15/pc): 0x2401d32c
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x20000380 (main)
E: Halting system
```

Also hello world gives no output
```
west build -p -b qemu_cortex_m0 samples/hello_world/ -- -DCONFIG_GEN_ISR_TABLES=n
```

In contrast `qemu_cortex_m3` (which is not a nordic board) **WORKS**

Another (probably related, but minor issue):

When running a test with `GEN_ISR_TABLES=n` with IAR toolchain:
```
 scripts/twister --force-toolchain -p qemu_cortex_m0 -T tests/arch/arm/arm_irq_vector_table/ -W
[...]
Error[Lc036]: no block or place matches the pattern "ro data section .intList in nrf_timer_timer.c.o"
Error[Lc036]: no block or place matches the pattern "ro data section .intList in clock_control_nrf.c.o"
Error[Lc036]: no block or place matches the pattern "ro data section .intList in gpio_nrfx.c.o"
```
seems to indicate that Nordic code is putting stuff in `.intList` even when `GEN_ISR_TABLES=n`

This error can be suppressed, and THIS particular test works, so it is a minor issue, but possibly related to the MAJOR problems above.

**Expected behavior**

A: Produce working code.
B: Not producing sections that are not placed anywhere by CMAKE_LINKER_GENERATOR (minor issue).

(Seems to be Nordic only? Other boards seem to work fine in this configuration)

nordic-piks, 2025-02-27:
@carlescufi can you help?

bjarki-andreasen, 2025-08-06:
Technically, CONFIG_GEN_ISR_TABLES can be used with ARM processors, but this would require the application and drivers to be rewritten to manually create and populate the IRQ vector table (the one at addr 0x0, pointed to by VTOR), zephyr's IRQ_CONNECT and IRQ_DIRECT_CONNECT will not work if `CONFIG_GEN_ISR_TABLES` is not enabled.