zephyrproject-rtos/zephyr #86348: esp32 use mcuboot espressif port. not boot zephyr app.

zhang-wenchao, 2025-02-26:
**Describe the bug**

I'm trying the mcuboot espressif port and can't boot the zephyr app.

**To Reproduce**

**Mcuboot:**

```
git clone https://github.com/mcu-tools/mcuboot
```

Modify the flash partition corresponding to the zephyr dts definition.

```

nano mcuboot/boot/espressif/port/esp32c3/bootloader.conf
```

```
CONFIG_ESP_FLASH_SIZE=4MB
CONFIG_ESP_BOOTLOADER_SIZE=0x20000

CONFIG_ESP_IMAGE_NUMBER=2

CONFIG_ESP_APPLICATION_SIZE=0x1C0000
CONFIG_ESP_IMAGE0_PRIMARY_START_ADDRESS=0x20000
CONFIG_ESP_IMAGE0_SECONDARY_START_ADDRESS=0x1E0000
CONFIG_ESP_IMAGE1_PRIMARY_START_ADDRESS=0x3A0000
CONFIG_ESP_IMAGE1_SECONDARY_START_ADDRESS=0x3A8000
CONFIG_ESP_SCRATCH_OFFSET=0x3E0000
CONFIG_ESP_SCRATCH_SIZE=0x1F000

CONFIG_ESP_BOOTLOADER_OFFSET=0x0000

CONFIG_ESP_MCUBOOT_WDT_ENABLE=y

```

zephyr.dts

```
...
		flash: flash-controller@60002000 {
			compatible = "espressif,esp32-flash-controller";
			reg = < 0x60002000 0x1000 >;
			#address-cells = < 0x1 >;
			#size-cells = < 0x1 >;
			flash0: flash@0 {
				compatible = "soc-nv-flash";
				erase-block-size = < 0x1000 >;
				write-block-size = < 0x4 >;
				reg = < 0x0 0x400000 >;
				partitions {
					compatible = "fixed-partitions";
					#address-cells = < 0x1 >;
					#size-cells = < 0x1 >;
					boot_partition: partition@0 {
						label = "mcuboot";
						reg = < 0x0 0x20000 >;
					};
					slot0_partition: partition@20000 {
						label = "image-0";
						reg = < 0x20000 0x1c0000 >;
					};
					slot1_partition: partition@1e0000 {
						label = "image-1";
						reg = < 0x1e0000 0x1c0000 >;
					};
					slot0_lpcore_partition: partition@3a0000 {
						label = "image-0-lpcore";
						reg = < 0x3a0000 0x8000 >;
					};
					slot1_lpcore_partition: partition@3a8000 {
						label = "image-1-lpcore";
						reg = < 0x3a8000 0x8000 >;
					};
					storage_partition: partition@3b0000 {
						label = "storage";
						reg = < 0x3b0000 0x30000 >;
					};
					scratch_partition: partition@3e0000 {
						label = "image-scratch";
						reg = < 0x3e0000 0x1f000 >;
					};
					coredump_partition: partition@3ff000 {
						label = "coredump";
						reg = < 0x3ff000 0x1000 >;
					};
				};
			};
		};
...
```

```
cmake -DCMAKE_TOOLCHAIN_FILE=tools/toolchain-esp32c3.cmake -DMCUBOOT_TARGET=esp32c3 -DESP_HAL_PATH=/home/life/Develop/platform/esp/esp-idf -DMCUBOOT_FLASH_PORT=/dev/ttyACM0 -B build -GNinja

ninja -C build/

ninja -C build/ flash

```

**Zephyr:**

```
cd samples/hello_world
nano prj.conf 
```

```
CONFIG_BOOTLOADER_MCUBOOT=y
```

```
west build -b esp32c3_devkitm -p always && west flash
```

**Expected behavior**

Start the zephyr app normally

**Impact**

Unable to start, unable to develop

**Logs and console output**

```
west espressif monitor
```

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c50e
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcdaa00,len:0x1720
load:0x403c7000,len:0x28d8
load:0x403d0000,len:0x2324
entry 0x403d22ce
[esp32c3] [INF] *** Booting MCUboot build v2.1.0-247-g7330df7c ***
[esp32c3] [INF] [boot] chip revision: v0.4
[esp32c3] [INF] [boot.esp32c3] SPI Speed      : 40MHz
[esp32c3] [INF] [boot.esp32c3] SPI Mode       : DIO
[esp32c3] [INF] [boot.esp32c3] SPI Flash Size : 4MB
[esp32c3] [INF] [boot] Enabling RNG early entropy source...
[esp32c3] [INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32c3] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32c3] [INF] Boot source: primary slot
[esp32c3] [INF] Image index: 0, Swap type: none
[esp32c3] [INF] Primary image: magic=bad, swap_type=0x1, copy_done=0x2, image_ok=0x2
[esp32c3] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32c3] [INF] Boot source: none
[esp32c3] [INF] Image index: 1, Swap type: none
[esp32c3] [ERR] Unable to find bootable image
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK
 - ebcc1b71af6b015fff787a581ec5895778647a3e

zhang-wenchao, 2025-02-26:
@rftafas

I really appreciate your guidance, but there are some problems.

I really hope you can help me see if this problem is a problem with mcuboot or zephyr. If it is a problem with mcuboot, I will transfer the issues to the upstream repository.

Thank you very much again.

closed 2025-02-26 by zhang-wenchao

rftafas, 2025-02-26:
Was this solved?

zhang-wenchao, 2025-02-26:
It worked, I had misunderstood the documentation. Thanks for your patience.