zephyrproject-rtos/zephyr #86377: tests: drivers: mbox: mbox_data: mimxrt11xx_evk: target broken

hakehuang, 2025-02-26:
**Describe the bug**

the commit 
```
commit aec0355380e20ba5ea4cdf98d476a7113a138673 (HEAD)
Author: Daniel DeGrasse <daniel@degrasse.com>
Date:   Fri Dec 6 16:56:13 2024 -0600

    boards: nxp: mimxrt11xx: change zephyr,flash node to ocram

    OCRAM region in Cortex-M4 memory map at 0x2020_0000 is simply an alias
    to the M4 TCM at 0x1FFE_0000, which the zephyr,flash node was previously
    set to. Using this base address for OCRAM allows the base address to
    match the one used in the M7 memory map, which simplifies loading the M4
    image from flash into RAM in the M7 init routine

    Signed-off-by: Daniel DeGrasse <daniel@degrasse.com>

:100644 100644 a92f93feb8d 475a166e787 M        boards/nxp/mimxrt1160_evk/mimxrt1160_evk_mimxrt1166_cm4.dts
:100644 100644 cbef2377dab 5cd416c9c73 M        boards/nxp/mimxrt1170_evk/mimxrt1170_evk_mimxrt1176_cm4.dts
```
makes the tests/drivers/mbox/mbox_data fail, console has no output.
if revert above commit and config the board(M4) with
```
CONFIG_BUILD_OUTPUT_HEX_GAP_FILL=y
```
the test can pass

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - mimxrt1170_evk and mimxrt1160_evk
 - What have you tried to diagnose or workaround this issue?
 - git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - aec0355380e20ba5ea4cdf98d476a7113a138673 

**To Reproduce**

```
west build -b mimxrt1160_evk/mimxrt1166/cm7 -p always --sysbuild tests/drivers/mbox/mbox_data/
west -vvvv flash --runner=jlink --id="721119608"
```

**Expected behavior**

Test can run and pass

**Impact**

real multi-core application

**Logs and console output**

when this happen, there is no console output

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.1.0-rc2-85-g8081008a08d

**Additional context**

This case is failed due to #82841 , but if we do the fix mentioned in above, we find a new issue caused by aec0355380e20ba5ea4cdf98d476a7113a138673

hakehuang, 2025-02-26:
@danieldegrasse

EmilioCBen, 2025-07-22:
Hello @hakehuang , I am not seeing this issue on my side testing with the mimxrt1170_evk, can you check again on this one?

hakehuang, 2025-07-23:
> Hello [@hakehuang](https://github.com/hakehuang) , I am not seeing this issue on my side testing with the mimxrt1170_evk, can you check again on this one?

this has been fixed, sorry forget to close it