zephyrproject-rtos/zephyr #86397: esp32s3: Fail to send packet when enable host name and mDNS responder

phb98, 2025-02-27:
**Describe the bug**
 When running wifi shell example with these config enable
```
CONFIG_NET_HOSTNAME_ENABLE=y    
CONFIG_NET_HOSTNAME="zephyr"        
CONFIG_MDNS_RESPONDER=y 
```
On the first "wifi scan" command, esp_wifi_drv indicate that it fail to send a packet. 
Not sure what the packet is. The return code is 0x3006(ESP_ERR_WIFI_STATE).
There is still scan result return.
**To Reproduce**
Run wifi shell example with those config above, then scan wifi. 

**Expected behavior**
No fail to send packet log

**Impact**
On my project, enable those configs lead to fatal crash when scanning wifi. To isolate i run this example 
and check if there were anything wrong with wifi. Looklike it has, but it does not crash like my project.

**Logs and console output**

here is the dump packet.
```
[00:00:02.006,000] <err> esp32_wifi: pkt to send 0x3fcaa6d0 len 46 buf 0x3fcb141a
[00:00:02.006,000] <err> esp32_wifi: Pkt data:
                                     01 00 5e 00 00 fb d8 3b  da 4d a9 a8 08 00 46 00 |..^....; .M....F.
                                     00 20 00 00 00 00 01 02  43 dd 00 00 00 00 e0 00 |. ...... C.......
                                     00 fb 94 04 00 00 16 00  09 04 e0 00 00 fb       |........ ......  
[00:00:02.006,000] <err> esp32_wifi: Failed to send packet:0x3006
```
**Environment (please complete the following information):**
Zephyr main branch, commit id: af2f497ad0e89e5dc99efc627040a7a7813a3e28

**Additional context**

sylvioalves, 2025-03-01:
@phb98, which soc?

sylvioalves, 2025-03-01:
@phb98 I think that is fixed also by this PR: #86517
The "Failed to send packet" happens when there is no connection and Wi-Fi tries to send packets.. Would you check that?

phb98, 2025-03-02:
@sylvioalves Hi, it's the esp32s3. I will give your PR a try

closed 2025-05-19 by github-actions[bot]