zephyrproject-rtos/zephyr #86399: tests/drivers/uart/uart_async_api/drivers.uart.async_api fails at nrf54h20

nordic-piks, 2025-02-27:
- **Describe the bug**
The tests:
- tests/drivers/uart/uart_async_api/drivers.uart.async_api

Observed for
- nrf54h20dk@0.9.0/nrf54h20/cpuapp
- nrf54h20dk@0.9.0/nrf54h20/cpurad

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf54h20dk@0.9.0/nrf54h20/cpuapp connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/drivers/uart/uart_async_api -p nrf54h20dk@0.9.0/nrf54h20/cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v4.1.0-rc2-57-g05fb46cb9177 ***
W: Init not done!
Running TESTSUITE uart_async_chain_read
===================================================================
UART instance:uart@9d6000
START - test_chained_read
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x2f0146c0
E: r0/a1:  0x0000004d  r1/a2:  0x2f01713c  r2/a3:  0x2f0146c1
E: r3/a4:  0x2f0146c0 r12/ip:  0x0e0a7c41 r14/lr:  0x0e0a7727
E:  xpsr:  0x81001800
E: Faulting instruction address (r15/pc): 0x0e0aa362
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x2f0148a0 (test_chained_read)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.17.0
 - Commit SHA or Version used: v4.1.0-rc2-57-g05fb46cb9177

nordic-piks, 2025-02-27:
Guessing that may be related to https://github.com/zephyrproject-rtos/zephyr/pull/85548

sswetha18, 2025-02-27:
Could you please try by disabling Userspace
CONFIG_TEST_USERSPACE = n

nordic-krch, 2025-02-27:
@sswetha18 you made all those buffers not only aligned but also `static` and because of that test is failing when userspace is enabled. Can you remove `static` from those variables? Test was supporting `CONFIG_TEST_USERSPACE=y` and it should stay that way.

sswetha18, 2025-02-27:
sure will remove

djiatsaf-st, 2025-02-27:
Hello @sswetha18, I have also encountered the same issue on many STM32 platforms.

The suggestion made by @nordic-krch should normally correct the regression.

sswetha18, 2025-02-28:
https://github.com/zephyrproject-rtos/zephyr/pull/86445

nordic-krch, 2025-03-04:
Fixed by #86445.

closed 2025-03-04 by nordic-krch