zephyrproject-rtos/zephyr #86409: NXP USB drivers don't work with usb_next

toonst, 2025-02-27:
**Describe the bug**

The next usb stack (`usb_next`) is not working on NXP. 

**To Reproduce**

`west build -p -b frdm_rw612 samples/subsys/usb/mass/ -- -DEXTRA_DTC_OVERLAY_FILE="ramdisk.overlay" -DCONF_FILE=usbd_next_prj.conf`

**Expected behavior**

Working USB that enumerates.

**Impact**

Can use the old usb stack, so no big impact.

**Logs and console output**
```
uart:~$ *** Booting Zephyr OS build v4.1.0-rc1-57-g50c21f159143 ***
[00:04:11.230,498] <inf> main: No file system selected
[00:04:11.230,592] <err> usbd_core: Failed to init device driver
[00:04:11.230,596] <err> usbd_sample_config: Failed to initialize device support
[00:04:11.230,599] <err> main: Failed to initialize USB device
[00:04:11.230,601] <err> main: Failed to enable USB
```

**Environment (please complete the following information):**

 - OS: MacOS
 - Toolchain (e.g Zephyr SDK, ...)
 - commit sha 50c21f15914

**Additional context**

wima88, 2025-04-02:
I observed something while troubleshooting this issue with USB initialization. The error seemed to stem from the driver being unable to access the `USBOTG` register (0x40145000). It looks like the following three registers were not properly set up and were still holding their default values:

1. PRSTCTL0 (0x40000010)
2. PSCCTL0 (0x40001010)
3. USB_CTRL (0x40003004)

Below is the GDB capture just before sample_usbd_init() was called:
```
Breakpoint 1, enable_usb_device_next () at /home/nuc/frdm-rw612/mass/src/main.c:63
63              sample_usbd = sample_usbd_init_device(NULL);
(gdb) x/xw 0x40000010
0x40000010:     0xf1500106
(gdb) x/xw 0x40001010
0x40001010:     0x02010e07
(gdb) x/xw 0x40003004
0x40003004:     0x0082aa80
(gdb) x/xw 0x50000010
0x50000010:     0xf1500106
(gdb) x/xw 0x50001010
0x50001010:     0x02010e07
(gdb) x/xw 0x50003004
0x50003004:     0x0082aa80
(gdb) x/xw 0x40145000
0x40145000:     0x00000000
(gdb) x/xw 0x50145000
0x50145000:     0x00000000
```

If I understand correctly, at this point, I should be able to access the USB base register and read the values. Can anyone point out where in the USB device initialization process these registers are supposed to be set up?

matt-wood-ct, 2025-04-08:
Yep this matches the problem I was having, good to see a fix is already in progress

closed 2025-04-10 by kartben