zephyrproject-rtos/zephyr #86412: Failed on compiling zephyr zigbee samples on babblesim board

Shangzewen, 2025-02-27:
Dear Author
I believe zephyr suppsoed to support zigbee samples but i could not located them in the main repo, i have to locate them on a different repo. 
On the other hand, i find out that as claim, that babblesim supposed to support 802.15.4 protocol which is zigbee,
however, when i trying to build zephyr zigbee light_bulb samples on bsim board, it gives me errors. I was wondering how should i make this simulation works on zigbee?

Shangzewen, 2025-02-27:
Dear @aescolar 
Thanks again for the help
I following the normal building procedure which list at the official website:
https://docs.zephyrproject.org/latest/develop/getting_started/index.html
and compiled babblesim by following the instruction mentioned in the following link:
https://docs.zephyrproject.org/latest/boards/native/nrf_bsim/doc/nrf52_bsim.html#nrf52-bsim
i am able to compile the babblesim project no problem.

Since i could not locate the zigbee samples in the original zephyr project, 
I tried to follow the following instructions which i found online:
mkdir -p ~/zephyrproject_zigbee/nrf
cd ~/zephyrproject_zigbee/nrf
unset ZEPHYR_BASE (Incase the base address was set previously)
west init -m https://github.com/nrfconnect/sdk-nrf
west update

Then i am able to found the zigbee samples under directory (the branch i used for sdk_nrf is 2.7)
zephyrproject_zigbee/nrf/nrf/samples/zigbee
It looks like the Zephyr project is messed up completely, I am able to build both bluetooth and zigbee samples on board nrf52840dk/nrf52840 but can not build them both on nrf52_bsim board. 

but when i trying to build the zigbee examples on board nrf52_bsim I encountered following issues:
1:
multiple header files are missing for compiling zigbee on bsim which looks like the include is not merged well in Cmake file 

[image]

After i manually added the dependencies, it gave me :
 

when i trying to build the bluetooth cnetral_hr samples which  located at:

I had following issues
1: 
  This combination of SoC and floating point ABI is notsupported by the
  SoftDevice Controller lib.
2: 
mpsl.h could not be found

[image]

All the errors I encountered for compiling Bluetooth_central_hr on bsim board does not occur if I just use a plain zephyr project without adding zigbee support which seems really weird.

more logs:
Loading Zephyr default modules (Zephyr base).
-- Application: /home/asset/zephyrproject_zigbee/nrf/zephyr/samples/bluetooth/central_hr
-- CMake version: 3.28.3
-- Found Python3: /home/asset/zephyrproject_zigbee/.venv/bin/python (found suitable version "3.12.3", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/asset/.cache/zephyr
-- Zephyr version: 3.7.99 (/home/asset/zephyrproject_zigbee/nrf/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: nrf52_bsim, qualifiers: native
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.17.0 (/home/asset/zephyr-sdk-0.17.0)
-- Found toolchain: host (gcc/ld)
-- Found Dtc: /home/asset/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/asset/zephyrproject_zigbee/nrf/zephyr/boards/native/nrf_bsim/nrf52_bsim.dts
-- Generated zephyr.dts: /home/asset/zephyrproject_zigbee/nrf/build/central_hr/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/asset/zephyrproject_zigbee/nrf/build/central_hr/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/asset/zephyrproject_zigbee/nrf/build/central_hr/zephyr/dts.cmake

warning: Experimental symbol BT_CTLR_SDC_BSIM_BUILD is enabled.

Parsing /home/asset/zephyrproject_zigbee/nrf/zephyr/Kconfig
Loaded configuration '/home/asset/zephyrproject_zigbee/nrf/zephyr/boards/native/nrf_bsim/nrf52_bsim_defconfig'
Merged configuration '/home/asset/zephyrproject_zigbee/nrf/zephyr/samples/bluetooth/central_hr/prj.conf'
Merged configuration '/home/asset/zephyrproject_zigbee/nrf/build/central_hr/zephyr/.config.sysbuild'
Configuration saved to '/home/asset/zephyrproject_zigbee/nrf/build/central_hr/zephyr/.config'
Kconfig header saved to '/home/asset/zephyrproject_zigbee/nrf/build/central_hr/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /usr/bin/ld.bfd (found version "2.42") 
-- The C compiler identification is GNU 13.3.0
-- The CXX compiler identification is GNU 13.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
-- Using BSIM from BSIM_COMPONENTS_PATH=/home/asset/zephyrproject_zigbee/nrf/zephyr/../tools/bsim/components/ BSIM_OUT_PATH=/home/asset/zephyrproject_zigbee/nrf/zephyr/../tools/bsim
-- Setting build type to 'MinSizeRel' as none was specified.
-- Using ccache: /usr/bin/ccache
CMake Error at /home/asset/zephyrproject_zigbee/nrf/nrfxlib/softdevice_controller/CMakeLists.txt:28 (message):
  This combination of SoC and floating point ABI is notsupported by the
  SoftDevice Controller lib.

-- Configuring incomplete, errors occurred!
CMake Error at cmake/modules/sysbuild_extensions.cmake:514 (message):
  CMake configure failed for Zephyr project: central_hr

Based on my suspect, It probably because i did not include the zigbee samples correctly since i could not find any tutorial online. After I re-init the zephyr project, the dependencies got messed up and causing all compiling on bsim failed

nordicjm, 2025-02-27:
Zigbee is not supported in zephyr

closed 2025-02-27 by nordicjm

aescolar, 2025-02-27:
> Zigbee is not supported in zephyr,

@Shangzewen and unfortunately, at this point, you cannot build targeting the nrf*bsim boards selecting the proprietary Soft Device Controller. And neither the zigbee library provided with NCS.
So this use case is just not supported.