zephyrproject-rtos/zephyr #86413: samples: sensor: accel_trig : fail on many stm32 platform

djiatsaf-st, 2025-02-27:
**Describe the bug**

The changes made in this PR https://github.com/zephyrproject-rtos/zephyr/pull/85939 introduce a bug. The sample fails on STM32 platforms that don't have the x_nucleo_iks01a3 shield on them.

**To Reproduce**

1. have STM32 board connected without x_nucleo_iks01a3 shied on it. 
2. go to your zephyr directory
3. ./scripts/twister  --device-testing --device-serial /dev/ttyACM0 -p nucleo_g071rb -T  samples/sensor/accel_trig
4. See handler.log file 

**Expected behavior**

Valid console output if board have shied on it or integrated sensor, ortherwise skip sample scenario.

**Logs and console output**

* Booting Zephyr OS build v4.1.0-rc2-70-gc135c37b3ad6 ***
Device lis2dw12@19 is not ready

**Environment (please complete the following information):**

- OS: Ubuntu 22.04 LTS
- Toolchain Zephyr SDK 0.17.0

**Additional context**

The sample has valid output on the stm32f3_disco since the sensor is integrated on it.

nordic-piks, 2025-02-28:
Found also with nrf platforms. Looks like this configuration is missing `fixture` which will select for testing only at devices which has shield mounted.

nordic-piks, 2025-02-28:
@pcurt FYI

pcurt, 2025-02-28:
This is because the depends on is too generic.
```
    depends_on:
      - arduino_i2c
```

I think we can add a filter and a fixture
```
--- a/samples/sensor/accel_trig/sample.yaml
+++ b/samples/sensor/accel_trig/sample.yaml
@@ -37,8 +37,10 @@ tests:
       - apard32690/max32690/m4
   sample.sensor.accel_trig.shield-tap:
     harness: console
+    filter: dt_compat_enabled("st,lis2dw12")
     tags: sensors
     harness_config:
+      fixture: fixture_sensor_accel_tap
       type: one_line
       regex:
         - "^\\s*[0-9A-Za-z_,+-.]*@[0-9A-Fa-f]* \\[m\/s\\^2\\]:    \
@@ -57,6 +59,7 @@ tests:
             dt_compat_enabled("st,iis2dlpc") or dt_compat_enabled("st,lis2dw12")
     tags: sensors
     harness_config:
+      fixture: fixture_sensor_accel_tap
       type: one_line
       regex:
         - "^\\s*[0-9A-Za-z_,+-.]*@[0-9A-Fa-f]* \\[m\/s\\^2\\]:    \
```

djiatsaf-st, 2025-03-06:
https://github.com/zephyrproject-rtos/zephyr/pull/86538 Merged

nordic-piks, 2025-03-07:
The issue still occurs at nrf52dk/nrf52832:
```
*** Booting Zephyr OS build v4.1.0-rc3-134-g3d0a7dab1a97 ***
Device lis2dw12@19 is not ready
```
Adding depends_on does not solve the issue, fixture will do that.

nordic-piks, 2025-05-30:
Issue still exist, there is missing fixture to indicate that specific HW is needed on the top of nrf52dk/nrf52832

nordic-piks, 2025-05-30:
Will be fixed by https://github.com/zephyrproject-rtos/zephyr/pull/90840

closed 2025-06-02 by nordic-piks