zephyrproject-rtos/zephyr #86435: http_get sample with TLS fails on NXP FRDM-MCXN947

DerekSnell, 2025-02-27:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/86373

Sample fails with this command using Zephyr tag `v4.1.0-rc2`
```
west build -b frdm_mcxn947/mcxn947/cpu0 -- -DEXTRA_CONF_FILE="overlay-tls.conf"
```

console output:
```
---- Opened the serial port /dev/ttyACM1 ----
[00:00:00.605,000] <inf> phy_mii: PHY (0) ID 80008000
*** Booting Zephyr OS build v4.1.0-rc2-86-g384fe7a05d7d ***
[00:00:00.606,000] <wrn> net_sock_tls: No entropy device on the system, TLS communication is insecure!
[00:00:05.108,000] <wrn> net_dhcpv4: DHCP server provided more DNS servers than can be saved
[00:00:05.109,000] <inf> net_dhcpv4: Received: 10.5.1.157
[00:00:05.208,000] <inf> http_get: Preparing HTTP GET request for http://google.com:443/
[00:00:05.209,000] <inf> http_get: getaddrinfo status: 0
[00:00:05.209,000] <inf> http_get: addrinfo @0x300247a8: ai_family=1, ai_socktype=1, ai_protocol=6, sa_family=1, sin_port=1bb
[00:00:05.209,000] <inf> http_get: sock = 5
[00:00:05.209,000] <inf> http_get: Connecting to server...
[00:00:08.217,000] <err> net_sock_tls: TLS handshake error: -0x7200
[00:00:08.218,000] <err> http_get: Error: -1
exit
```
and
```
[00:00:10.429,000] net_sock_tls: TLS handshake error: -0x7300
[00:00:10.431,000] net_sock_tls: TLS handshake error: -0x7200
```
Those errors refer to :
```
/** A message could not be parsed due to a syntactic error. /
#define MBEDTLS_ERR_SSL_DECODE_ERROR -0x7300
/* An invalid SSL record was received. */
#define MBEDTLS_ERR_SSL_INVALID_RECORD -0x7200
```

kartben, 2025-03-04:
@jukkar @rlubos could you please take a look? Is it platform specific or a broader issue?

rlubos, 2025-03-05:
The sample worked just fine for me (on `native_sim`, `qemu_x86`), at least at my location.

I suggest enabling mbed TLS logs, they're usually pretty good - the required configs are commented out in the TLS overlay file (might be needed to increase `CONFIG_LOG_BUFFER_SIZE` to avoid packet drops).

I recall from the past seeing those mbed TLS bad record in case configured record size was too small, perhaps try increasing `CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN` - might need to increase `CONFIG_MBEDTLS_HEAP_SIZE` as well.

sorru94, 2025-03-07:
> The sample worked just fine for me (on `native_sim`, `qemu_x86`), at least at my location.
>
> I suggest enabling mbed TLS logs, they're usually pretty good - the required configs are commented out in [...]

This issue is specific to the NXP FRDM MXCN947 board. The sample works fine on other boards (for example I can run it on the NXP i.MX RT1064 EVK). 
You can refer to the discussion mentioned in the description for more information on how to reproduce the issue (#86373).

The values of `CONFIG_MBEDTLS_HEAP_SIZE` and `CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN` are already quite high in the TLS overlay (60000 and 7168 respectively).
I increased them additionally up to  `CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=32768` and `CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=120000` but the issue persists.

sorru94, 2025-05-07:
This issue still needs fixing as of my knowledge.

PR #91312 (2025-06-11): eth_nxp_enet_qos fixes on MCXN947
This PR fixes the following in ENET QoS driver:

- out of order processing of DMA-prepared buffers (previously started from index 0, now from the next position prepared by DMA in the ring of descriptors)
- receiving of frames split into multiple buffers (previously dropped, now net_pkt is composed of multiple buffers if frame is longer than single buffer size now and passed to network stack)
- increases the default configuration for number of buffers, so that at least 1 frame of maximum size can be received (from 4 to 16)

Together with https://github.com/zeph [... cut]
closed 2025-06-11 by dkalowsk