zephyrproject-rtos/zephyr #86477: mDNS resolver doesn't work if IPv4 is disabled

kevinior, 2025-02-28:
**Describe the bug**

The mDNS resolver always times out if only IPv6 is enabled, even if there has been a valid response to the query.

I have tested this with both our own hardware (STM32H743-based) and the ST Nucleo-H563ZI. The problem is present in both Zephyr 3.7.1 and current main (commit 95c1c850bcda13ff6ebc187039337b56356cc78c).

I tried enabling more debug output but it seems like the response packet is being ignored at a low level because I couldn't see any indication of it being received.

**To Reproduce**

Configure the samples/net/dns_resolve sample with IPv4 disabled, here are the diffs in my samples/net/dns_resolve/prj.conf:

```diff
diff --git a/samples/net/dns_resolve/prj.conf b/samples/net/dns_resolve/prj.conf
index de837ae0828..4c90263efbf 100644
--- a/samples/net/dns_resolve/prj.conf
+++ b/samples/net/dns_resolve/prj.conf
@@ -11,7 +11,7 @@ CONFIG_NET_LOG=y
 CONFIG_LOG=y
 
 CONFIG_NET_IPV6=y
-CONFIG_NET_IPV4=y
+CONFIG_NET_IPV4=n
 CONFIG_NET_DHCPV4=n
 
 CONFIG_ZVFS_POLL_MAX=5
@@ -33,8 +33,8 @@ CONFIG_DNS_SERVER_IP_ADDRESSES=y
 CONFIG_DNS_NUM_CONCUR_QUERIES=5
 
 # Use local dnsmasq server for testing
-CONFIG_DNS_SERVER1="[2001:db8::2]:15353"
-CONFIG_DNS_SERVER2="192.0.2.2:15353"
+CONFIG_DNS_SERVER1="[fe80::a92:4ff:fe49:be3b]:15353"
+# CONFIG_DNS_SERVER2="192.0.2.2:15353"
 
 # Google DNS IPv4 and IPv6 servers
 #CONFIG_DNS_SERVER1="8.8.8.8"
@@ -50,10 +50,10 @@ CONFIG_NET_SHELL=y
 
 CONFIG_NET_CONFIG_SETTINGS=y
 CONFIG_NET_CONFIG_NEED_IPV6=y
-CONFIG_NET_CONFIG_NEED_IPV4=y
+# CONFIG_NET_CONFIG_NEED_IPV4=y
 CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
-CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
-CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
-CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
+CONFIG_NET_CONFIG_PEER_IPV6_ADDR="fe80::a92:4ff:fe49:be3b"
+# CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
+# CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
 
 CONFIG_MAIN_STACK_SIZE=1504
```

Build: `west build -p -b nucleo_h563zi dns_resolve/`
and flash `west flash`

Connect the board directly to a Linux PC and run mDNS responders for IPv4 and IPv6 addresses: `avahi-publish-address -R ferroamp.local fe80::a92:4ff:fe49:be3b` and `avahi-publish-address -R 192.0.2.2`

**Expected behavior**

mDNS queries are resolved:
```
uart:~$ net dns query ferroamp.local
Query for 'ferroamp.local' sent.
dns: 192.0.2.2
dns: All results received
uart:~$ net dns query ferroamp.local AAAA
IPv6 address type
Query for 'ferroamp.local' sent.
dns: fe80::a92:4ff:fe49:be3b
dns: All results received
```

**Actual behavior**

mDNS queries time out:
```
uart:~$ net dns query ferroamp.local
Query for 'ferroamp.local' sent.
dns: Timeout while resolving name.
uart:~$ net dns query ferroamp.local AAAA
IPv6 address type
Query for 'ferroamp.local' sent.
dns: Timeout while resolving name.
```

**Impact**

Our system is intended to run in an IPv6-only environment, so this is a significant annoyance.

**Logs and console output**

Wireshark captures attached: mdns_ok.pcapng, mdns_fail.pcapng

[mdns.zip](https://github.com/user-attachments/files/19027613/mdns.zip)

**Environment (please complete the following information):**

 - OS: Ubuntu 24.04
 - Toolchain: zephyr-sdk-0.16.5
 - Commit SHA: 9f824289b28d7aea2eee74f62787c385a5005453 (tag v3.7.1)
 - Commit SHA: 95c1c850bcda13ff6ebc187039337b56356cc78c (main)

jukkar, 2025-02-28:
I could replicate the issue. The device should be able to receive the DNS response if you set `CONFIG_MDNS_RESPONDER=y`.
It seems that we do not join `ff02::fb` multicast group if mDNS responder is not set, which means that the mDNS response sent by avahi is ignored. We should probably join the group even if the mDNS responder is not activated, I will investigate this more.

kevinior, 2025-03-03:
Thanks for the quick response @jukkar .  Yes, I can confirm that enabling the mDNS responder works. I guess it works for IPv4 because Zephyr always joins the 224.0.0.1 multicast group by default.

This will be OK for us as a workaround for now, although I'd rather not bring in a whole extra thing listening to the network.

jukkar, 2025-03-03:
Yes, my suggestion for enabling responder was just a workaround, we should support this without needing to enable the responder.

jukkar, 2025-03-04:
@kevinior I have a fix proposal here #86617, please try and check if it works for your use case.

kevinior, 2025-03-04:
I don't have the hardware with me today but I'll test it first thing tomorrow.

commit d955af4e (2025-03-07): net: dns: Join mDNS multicast group for resolving

If mDNS resolver is enabled but mDNS responder is not, then
mDNS multicast address group is not joined. This would prevent
the mDNS resolver to receive the responses. Fix this by
joining the mDNS multicast group if mDNS responder is not
enabled (because the responder will join the group itself).

Fixes #86477
closed 2025-03-07 by aescolar