zephyrproject-rtos/zephyr #86480: *native_posix* driver name renaming

aescolar, 2025-02-28:
We have quite a few drivers with `native_posix` in their name, DTS and/or kconfig options and file names.
Now that native_posix is being removed we have two options: We let them be, or we rename them
following appropriate deprecation.
Letting them be as they are could be a bit confusing for users, but minimizes churn, and viceversa.
We also don't necessarily need to rename all.

### Possible renaming:

* `native_sim` board specific drivers:
  * native_posix_timer => native_sim_timer 
     * CONFIG_NATIVE_POSIX_TIMER => CONFIG_NATIVE_SIM_TIMER
  * counter_native_posix => counter_native_sim
* native simulator specific (can be used in all current POSIX arch targets):
  * fake_entropy_native_posix => fake_entropy_native_sim
    * zephyr_native_posix_rng => zephyr_native_sim_rng
    * CONFIG_FAKE_ENTROPY_NATIVE_POSIX => CONFIG_FAKE_ENTROPY_NATIVE_SIM
  * CONFIG_UART_NATIVE_POSIX => CONFIG_UART_NATIVE_PTTY
  * CONFIG_ETH_NATIVE_POSIX => CONFIG_ETH_NATIVE_TUN_TAP ?
* Any posix arch target:
  * log_backend_native_posix => 
  * usb_native_posix ?
* DTS zephyr,native-posix-cpu => zephyr,native-sim-cpu ?

aescolar, 2025-02-28:
@jukkar @rlubos @nordic-krch @ceolin @jfischer-no @tmon-nordic @dcpleung , input is welcome

tmon-nordic, 2025-02-28:
`USB_NATIVE_POSIX` will be depreated alongside the old USB stack. The implementation is completely wrong. The new stack has better USBIP support which does not share the `USB_NATIVE_POSIX` problems (including, but not limited to confusing name).

jukkar, 2025-02-28:
No objection for changing native_posix -> native_sim in Ethernet driver. The driver has not seen much changes lately so I am not expecting merge issues in the future. Renaming Kconfig options is of course a bigger change that might affect users but I suppose that is unavoidable.

ceolin, 2025-02-28:
@aescolar No objections on my side for the entropy as well :)

closed 2025-05-05 by aescolar