zephyrproject-rtos/zephyr #86488: STM32: Support for QSPI PSRAM on STM32H5?

kristoftjonck, 2025-02-28:
In STM32CubeMX for the STM32H567, it is possible to use QSPI to extend memory with PSRAM, such as the IS66WVS1M8ALL. However, I couldn't find direct support for this functionality in Zephyr.

Are there any existing samples or examples demonstrating how to integrate QSPI PSRAM in Zephyr? If not, what would be the recommended approach to enable this functionality?

erwango, 2025-03-03:
Adding @gautierg-st who has done a similar work recently on STM32N6 and can show the way for H5.

erwango, 2025-03-03:
Here it is actually: https://github.com/zephyrproject-rtos/zephyr/pull/85564

erwango, 2025-06-30:
A similar work: https://github.com/zephyrproject-rtos/zephyr/pull/92381

mariopaja, 2025-11-02:
@kristoftjonck In case this is still an open issue for you. I have implemented APS6404 (QSPI) in XSPI bus. I plan to test it soon on OSPI Bus