zephyrproject-rtos/zephyr #86499: net: Critical Mutex Deadlock

legoabram, 2025-02-28:
**Describe the bug**
Running on an STM32H753VG with ethernet, there is a condition where the `rx_q[0]` thread and `sys_work_q` running `rs_timeout` can get into a mutex deadlock. The work queue acquires the iface mutex, and the rx_q acquires the iface TX mutex, and then they both try to acquire the other mutex. See the call stacks and iface provided for more details.

**To Reproduce**
Unfortunately, given the nature of the issue, I don't really have a way to replicate this easily. I do have core dumps of a failed system however, so I can provide any information needed at any time.

**Expected behavior**
No mutex deadlock?

**Impact**
This is a massive showstopper. This disables our primary functionality in a way that our WDT can't detect. And since it locks up the work queue as well, it breaks several other operations our device needs. We can still recover failed prototype devices in the field thanks to additional debug mechanisms, but we can't go into production with this bug in place.

**Logs and console output**

[image]

[image]

[image]

**Environment**

 - OS: Windows 11
 - Toolchain Zephyr SDK 0.17
 - Zephyr: 064fcfca6b4425c1481d571ec6acf8778f30b6d6

legoabram, 2025-02-28:
It's possible that this issue has already been addressed in a newer version of Zephyr, but we can't afford the time right now to upgrade if we don't know for certain it will fix the problem.

jukkar, 2025-03-03:
The commit indicates this is 4.0.0-rc2

jukkar, 2025-03-03:
@legoabram thanks for excellent analysis of the issue. I sent a fix proposal for review here #86553, could you try it if it fixes the issue you are seing.

legoabram, 2025-03-03:
@jukkar Thank you for the kind words, and more so the potential fix! Unfortunately, this is a fickle, race condition issue, so I can't guarantee an answer. I'll push to have the fix implemented on the particular product most affected, and report if it crashes.

PR #86553 (2025-03-05): net: if: Release the interface lock early in IPv6 RS timeout handler
The net_if.c:rs_timeout() is sending a new IPv6 router solicitation message to network by calling net_if_start_rs(). That function will then acquire iface->lock and call net_ipv6_start_rs() which will try to send the RS message and acquire TX send lock.
During this RS send, we might receive TCP data that could try to send an ack to peer. This will then in turn cause also TX lock to be acquired. Depending on timing, the lock ordering between rx thread and system workq might mix which could lead to deadlock. Fix this issue by r [... cut]
commit 4926698b (2025-03-05): net: if: Release the interface lock early in IPv6 RS timeout handler

The net_if.c:rs_timeout() is sending a new IPv6 router solicitation
message to network by calling net_if_start_rs(). That function will
then acquire iface->lock and call net_ipv6_start_rs() which will try
to send the RS message and acquire TX send lock.
During this RS send, we might receive TCP data that could try to
send an ack to peer. This will then in turn cause also TX lock
to be acquired. Depending on timing, the lock ordering between
rx thread and system workq might mix which could lead to deadlock.
Fix this issue by  [... cut]
commit 9d55a601 (2025-04-01): net: if: Release the interface lock early in IPv6 RS timeout handler

The net_if.c:rs_timeout() is sending a new IPv6 router solicitation
message to network by calling net_if_start_rs(). That function will
then acquire iface->lock and call net_ipv6_start_rs() which will try
to send the RS message and acquire TX send lock.
During this RS send, we might receive TCP data that could try to
send an ack to peer. This will then in turn cause also TX lock
to be acquired. Depending on timing, the lock ordering between
rx thread and system workq might mix which could lead to deadlock.
Fix this issue by  [... cut]
PR #86734 (2025-04-01): [Backport v3.7-branch] net: if: Release the interface lock early in IPv6 RS timeout handler
This is a backport of https://github.com/zephyrproject-rtos/zephyr/pull/86553 to v3.7. It does not have commit d5335d30c12daa2524a40690b7863f1df0299f54 as the feature it is fixing was not present in v3.7

In order to avoid any mutex deadlocks between iface->lock and TX lock, release the interface lock before calling a function that will acquire TX lock. See previous commit for similar issue in RS timer handling. So here we create a separate list of multicast addresses that are to be rejoined when networ [... cut]