zephyrproject-rtos/zephyr #86552: tests: tests/arch/arm/arm_irq_vector_table: mimxrt1180_evk/mimxrt1189/cm33: failed: Timeout

hakehuang, 2025-03-03:
**Describe the bug**
below case are failed
arch.arm.irq_vector_table.vector_table.arm_irq_vector_table

test is failed on v4.1.0-rc3 on mimxrt1180_evk/mimxrt1189/cm33
testcase path is tests/arch/arm/arm_irq_vector_table

see logs for details

**To Reproduce**
1. 
```
scripts/twister -p mimxrt1180_evk/mimxrt1189/cm33  -T tests/arch/arm/arm_irq_vector_table -s  arch.arm.irq_vector_table
```
or
```
# west twister -p mimxrt1180_evk/mimxrt1189/cm33 -T tests/arch/arm/arm_irq_vector_table -s -test arch.arm.irq_vector_table
```
2. See error

```
Timeout
```

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux ): posix
 - Toolchain (e.g Zephyr SDK): zephyr
 - Commit SHA or Version used: v4.1.0-rc3

hakehuang, 2025-03-05:
> Could you help re-test on RT1180 platform?

it still not working for me. see below attached log

[handler.log](https://github.com/user-attachments/files/19084850/handler.log)

nashif, 2025-03-05:
it would help to add the log or any output in the bug report, so instead of an empty section like this:

[image]

it helps alot to have something like this instead of attaching log files (also please rename the log to a txt file so it can be opened directly in the browser):

```
Running TESTSUITE vector_table
===================================================================
START - test_arm_irq_vector_table
Test Cortex-M IRQs installed directly in the vector table
isr1 ran!
isr1 ran!
isr1 ran!
isr1 ran!
isr1 ran!
isr1 ran!
isr1 ran!
isr1 ran!
isr1 ran!
isr1 ran!
```

nashif, 2025-03-05:
looks like this hardware needs some special handling in the test? look at the test code, it does have some hardware specidic definitions.

lucien-nxp, 2025-03-06:
Hi @hakehuang ,
I checked the issue again today, this issue maybe caused by linkserver tool because there is no issue if your use hardware reset after download. I will involved linkserver owner for help.
Thank you.

closed 2025-05-20 by github-actions[bot]