zephyrproject-rtos/zephyr #86561: VIRTIO support in Zephyr

fkokosinski, 2025-03-03:
## Introduction

This RFC introduces VIRTIO[1] support in Zephyr RTOS along with the roadmap of future improvements.

### Problem description

VIRTIO is a communication protocol designed for paravirtualized devices interfacing. Adding support for VIRTIO in Zephyr allows Zephyr to use a wider array of devices in a virtualized environment. VIRTIO does virtualization in a more efficient way compared to emulating physical devices by leveraging the fact that VIRTIO employs mechanisms, such as laying structures efficiently in memory, to speed up the virtualization process.

### Proposed change

Introduction of VIRTIO drivers, and drivers for virtualized VIRTIO-based devices.

## Detailed RFC

### Proposed change (Detailed)

* Define VIRTIO API that will allow operation on VIRTIO devices, such as:
  * Device initialization
  * Feature flags/bits negotiation
  * Accessing virtqueues
  * Accessing device-specific data
* Implement drivers for VIRTIO-based devices methods using the VIRTIO API for transfer methods that are as of date supported in Zephyr: PCI and MMIO
* Implement drivers for specific VIRTIO-based devices, such as a filesystem device, a network device etc.

### Roadmap

* [ ] Define VIRTIO API: https://github.com/zephyrproject-rtos/zephyr/pull/83892
* [ ] Implement virtqueues: https://github.com/zephyrproject-rtos/zephyr/pull/83892
* [ ] Implement driver for VIRTIO PCI: https://github.com/zephyrproject-rtos/zephyr/pull/83892
* [ ] Implement driver for VIRTIO MMIO
* [ ] Implement non-mandatory VIRTIO optimizations like `VIRTIO_F_INDIRECT_DESC`
* [ ] Add support for legacy (pre v1.0 version of specification) devices
* [ ] Write drivers for VIRTIO devices, like:
  * [ ] Filesystem device
  * [ ] Network device
  * [ ] Block device
  * [ ] Entropy source
  * [ ] Console

### Dependencies/Limitations

Some of the VIRTIO device classes may not yet have suitable existing Zephyr APIs. Those APIs would have to be introduced along with them. The full list of device types can be found in [chapter 5](https://docs.oasis-open.org/virtio/virtio/v1.3/csd01/virtio-v1.3-csd01.html#x1-2330005) of the VIRTIO specification.

### Concerns and Unresolved Questions

* Not all non-mandatory VIRTIO features are as of date supported in QEMU

## References

[1] [Virtual I/O Device (VIRTIO) Version 1.3](https://docs.oasis-open.org/virtio/virtio/v1.3/virtio-v1.3.html)

fkokosinski, 2025-03-03:
Hey @rruuaanng, this item (VIRTIO in Zephyr) has been already discussed on the last Arch WG (February 25th), and this RFC is the result of this discussion:
* https://github.com/zephyrproject-rtos/zephyr/pull/83892#issuecomment-2682566853
* https://lists.zephyrproject.org/g/devel/topic/architecture_wg_minutes/111487964

I'm of course happy to discuss this further, but I believe it was decided last week that this requires no further Arch WG attention.

Xaloqi, 2026-10-25:
For the device list on this roadmap: there is now an out-of-tree virtio-can driver (VIRTIO 1.4 CAN device, ID 36) built on the in-tree virtio API and the standard Zephyr CAN API: https://github.com/Xaloqi/zephyr-virtio-can (Apache-2.0, experimental 0.1.0). It is unit-tested on qemu_cortex_a53 with a fake transport, and has run end-to-end as a Xen guest against rust-vmm's vhost-device-can. Known limits: no grant-DMA, bus-off detected on TX failure only (4.4.1's transports deliver no config-change interrupts). If a CAN device fits this roadmap, feedback on the driver is welcome.

fkokosinski, 2026-10-26:
Hey @Xaloqi - I think it does :) Feel free to submit a PR with the driver.

Xaloqi, 2026-10-26:
Hi @fkokosinski, thanks 

We are glad it fits. One constraint on our side: Zephyr's DCO needs a personal legal name on each commit, and Xaloqi contributes as an organisation only (We could explain the reasons, but not here), so we can't open the PR ourselves. Two routes work for us:

	1.	Someone carries it in under DCO (b). It's Apache-2.0, and the headers keep the Xaloqi copyright. We'll review, test it on our Xen setups and answer review comments on that PR, and keep fixing it in our repo.
	2.	It stays an out-of-tree module, and the RFC's driver list links it.

If you or anyone in the virtio work is up for (1), tell us what you'd want changed before it goes in-tree. The open points we know of: no config-change interrupts yet, and the heap requirement that #121017 would remove.