zephyrproject-rtos/zephyr #86596: nRF700x: More specific heap requirement documentation

JordanYates, 2025-03-04:
**Is your enhancement proposal related to a problem? Please describe.**

The existing defaults for `HEAP_MEM_POOL_ADD_SIZE_NRF70` are very barebones, providing little guidance on where the heap is being used, and how the value can be modified for different use cases.

https://github.com/zephyrproject-rtos/zephyr/blob/5736aedd022eba46863613989ece4ac442712976/drivers/wifi/nrf_wifi/Kconfig.nrfwifi#L69-L73

The current NCS tree of `samples/wifi` contains values ranging from `25'000` to `260'000`.
The current default of `150'000` doesn't occur in the NCS tree at all.

Burning `150'000` bytes on heap when the currently available Nordic flagship products (nRF53, nRF54L) only have 256 kB RAM total (57%) deserves more explanation than: `"Use a maximum that works for typical use cases and boards"`.

**Describe the solution you'd like**

Improved documentation on the choice symbol, ideally some pre-configured choice options that setup sane defaults for the various configuration options:

```
choice NRF70X_DEFAULT_CONFIG
    prompt "Baseline configuration"

config NRF70X_SCAN_ONLY

config NRF70X_LOW_MEMORY

config NRF70X_MAXIMUM_THROUGHPUT

endchoice

config HEAP_MEM_POOL_ADD_SIZE_NRF70
    default 25000 if NRF70X_SCAN_ONLY
    default 80000 if NRF70X_LOW_MEMORY
    default 150000 if NRF70X_MAXIMUM_THROUGHPUT
    ...

config NET_PKT_RX_COUNT
   default X if NRF70X_LOW_MEMORY
   default Y if NRF70X_MAXIMUM_THROUGHPUT
   ...

etc
```

JordanYates, 2025-04-09:
@jukkar @krish2718

JordanYates, 2025-04-11:
Drilling down a bit deeper into the problem, the size of the required additional system heap appears to be mostly driven by `NRF70_RX_NUM_BUFS * NRF70_RX_MAX_DATA_SIZE` and `NRF70_MAX_TX_TOKENS * NRF70_TX_MAX_DATA_SIZE`. Needing to recompute `HEAP_MEM_POOL_ADD_SIZE_NRF70` seems like an unnecessary extra step, can't the memory be statically allocated somewhere?

krish2718, 2025-04-11:
> Drilling down a bit deeper into the problem, the size of the required additional system heap appears to be mostly driven by `NRF70_RX_NUM_BUFS * NRF70_RX_MAX_DATA_SIZE` and `NRF70_MAX_TX_TOKENS * NRF70_TX_MAX_DATA_SIZE`. 

Yes, plus a fixed amount for non-pkt heap in data path.

> Needing to recompute `HEAP_MEM_POOL_ADD_SIZE_NRF70` seems like an unnecessary extra step, can't the memory be statically allocated somewhere?

IIUC, you are suggesting not to use Zephyr shared heap but just use those TX/RX macros and define driver heap? BTW, the latest driver moves away from shared heap and uses dedicated pools. https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/wifi/nrf_wifi/Kconfig.nrfwifi#L564 we can remove the `CONFIG_NRF_WIFI_DATA_HEAP_SIZE` and use the arithmetic above.

JordanYates, 2025-04-11:
> IIUC, you are suggesting not to use Zephyr shared heap but just use those TX/RX macros and define driver heap? BTW, the latest driver moves away from shared heap and uses dedicated pools. [...]

So I had two concerns with my suggestion, both of which seem to be addressed already:

 1. Memory couldn't be "shared" with other users while the WiFi interface is down

If you're transitioning to a dedicated memory pool anyway, this concern has already been addressed, since other users can't malloc out of it anyway. And it seems like a fragile use-case anyway.

 2. How to compute the data path memory in Kconfig (since Kconfig doesn't support + or *)

If you were sticking with the system heap, you would need to shove the result of a computation into a Kconfig symbol to be scooped up by the heap size calculation. But if you're no longer using the system heap, that is no longer true.

Since you've already gone to the effort of using a dedicated heap, it seems silly to me to require the user to "guess" the correct size through a Kconfig option, when you could just compute the needed size from the various Kconfig options at the location you declare the heap.

It would make it MUCH simpler for optimising RAM usage. Decrease `NUM_RX_BUFFERS`, RAM usage goes down. No more messing with heap size configs.

krish2718, 2025-04-11:
> It would make it MUCH simpler for optimising RAM usage. Decrease `NUM_RX_BUFFERS`, RAM usage goes down. No more messing with heap size configs.

Agreed, we did discuss this in the past, but due to `(since Kconfig doesn't support + or *)` we had dropped this idea. We will revisit and see if we can now implement the arithmetic in C directly, one less Kconfig for the user to worry about, thanks.