zephyrproject-rtos/zephyr #86600: modules: hostap: crypto requirements cleanup

JordanYates, 2025-03-04:
**Is your enhancement proposal related to a problem? Please describe.**

`WIFI_NM_WPA_SUPPLICANT_CRYPTO_ALT` claims to be enabling crypto support for the `hostap` module, but in reality is also seems to be taking over the job of `WIFI_NM_WPA_SUPPLICANT_CRYPTO_ENTERPRISE`.

Compare the two symbols:
https://github.com/zephyrproject-rtos/zephyr/blob/5736aedd022eba46863613989ece4ac442712976/modules/hostap/Kconfig#L125-L148

https://github.com/zephyrproject-rtos/zephyr/blob/5736aedd022eba46863613989ece4ac442712976/modules/hostap/Kconfig#L185-L191

Many selections on `WIFI_NM_WPA_SUPPLICANT_CRYPTO_ALT` appear to only be used for WPA2-Enterprise support.
This is supported by commits that are explicitly saying so (58b6bd66, 4bab37e7).

IMO Enterprise support should be enabled by `WIFI_NM_WPA_SUPPLICANT_CRYPTO_ENTERPRISE`, not the default crypto selection `WIFI_NM_WPA_SUPPLICANT_CRYPTO_ALT`.

**Describe the solution you'd like**

A cleanup of the crypto options such that the user specifies which authentication methods they want supported, and the appropriate options are enabled.

```
config WIFI_NM_WPA_SUPPLICANT_CRYPTO_PSK
   bool "WPA2-PSK support"
   select MBEDTLS_aa
   select MBEDTLS_bb

config WIFI_NM_WPA_SUPPLICANT_CRYPTO_ENTERPRISE_BASE
   bool "WPA2-Enterprise and WPA3-Enterprise support (minimal)"
   select MBEDTLS_xx
   select MBEDTLS_yy

config WIFI_NM_WPA_SUPPLICANT_CRYPTO_ENTERPRISE_ALL
   bool "WPA2-Enterprise and WPA3-Enterprise support (all ciphers)"
   select WIFI_NM_WPA_SUPPLICANT_CRYPTO_ENTERPRISE_BASE
   select MBEDTLS_ECP_ALL_ENABLED
   select MBEDTLS_KEY_EXCHANGE_ALL_ENABLED
```

JordanYates, 2025-04-04:
As an example, on Zephyr v4.1.0 I am able to connect to my home Wi-Fi network with only
```
config WIFI_NM_WPA_SUPPLICANT_CRYPTO_ALT
	bool "Crypto Mbedtls alt support for WiFi"
	select MBEDTLS
	select MBEDTLS_CIPHER_AES_ENABLED
	select MBEDTLS_CMAC
	select MBEDTLS_PKCS5_C
	select MBEDTLS_PK_WRITE_C
	select MBEDTLS_NIST_KW_C
	select MBEDTLS_HKDF_C
```
[image]

Compared to the upstream definition of
https://github.com/zephyrproject-rtos/zephyr/blob/f289e47ca5aec0b8f0c95b45a23862183c5832dd/modules/hostap/Kconfig#L128-L151

The reduced feature set saves just over 44kB of ROM.

krish2718, 2025-04-04:
Thanks @JordanYates this was raised in last Wi-Fi biweekly telco too, I will organize these configs into respective features, we definitely need ROM improvements after the last hostap upmerge.