zephyrproject-rtos/zephyr #86608: LE Audio: Update BASS / Scan Delegator to v.1.0.1

Thalley, 2025-03-04:
**Is your enhancement proposal related to a problem? Please describe.**
The Broadcast Audio Scan Service v.1.0.1 has been released and require a few changes in Zephyr to be compliant.

**Describe the solution you'd like**
Specifically erratum [26529](https://bluetooth.atlassian.net/jira/software/c/projects/ES/issues/ES-26529) needs to be implemented, which changes how the bad broadcast code is handled. 

The change required is that instead of copying the bad broadcast code to the bad broadcast code field, the spec now requires setting the bad broadcast code field to 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF (all octets set to 0xFF). 

**Describe alternatives you've considered**
N/A

**Additional context**
https://www.bluetooth.com/specifications/specs/broadcast-audio-scan-service-2/

gudipudiramanakumar, 2025-03-31:
Hey can I solve this issue

Thalley, 2025-03-31:
> Hey can I solve this issue

Sounds great! Let me know if you need any help in preparing a PR or understanding the current implementation. Make sure to check out https://docs.zephyrproject.org/latest/contribute/index.html if you are unfamiliar with the Zephyr contribution process

gudipudiramanakumar, 2025-04-02:
Hi @Thalley, I'm unable to access [26529](https://bluetooth.atlassian.net/jira/software/c/projects/ES/issues/ES-26529) can you guide me through how to.
And regarding the issue, I see that the below code needs to be changed, can you verify if this is the right approach

https://github.com/zephyrproject-rtos/zephyr/blob/43acb93607c2161608742c59ab98aa1d77c207e5/subsys/bluetooth/audio/bap_broadcast_assistant.c#L283-L285
to
```suggestion
	broadcast_code = 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF;
	(void)memcpy(recv_state->bad_code, broadcast_code,
		     sizeof(recv_state->bad_code));
```
and
https://github.com/zephyrproject-rtos/zephyr/blob/43acb93607c2161608742c59ab98aa1d77c207e5/subsys/bluetooth/audio/bap_scan_delegator.c#L952-L955
to
```suggestion
	broadcast_code = 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF;

	(void)memcpy(internal_state->broadcast_code, broadcast_code,
		     sizeof(internal_state->broadcast_code));
```

Thalley, 2025-04-02:
> Hi [@Thalley](https://github.com/Thalley), I'm unable to access [26529](https://bluetooth.atlassian.net/jira/software/c/projects/ES/issues/ES-26529) can you guide me through how to. And regarding the issue, I see that the below code needs to be changed, can you verify if this is the right approach
> 
> [zephyr/subsys/bluetooth/audio/bap_broadcast_assistant.c](https://github.com/zephyrproject-rtos/zephyr/blob/43acb93607c2161608742c59ab98aa1d77c207e5/subsys/bluetooth/audio/bap_broadcast_assistant.c#L283-L285)
> 
> Lines 283 to 285 in [43acb93](/zephyrproject-rtos/zephyr/commit/43acb93607c2161608742c59ab98aa1d77c207e5)
>  broadcast_code = net_buf_simple_pull_mem(&buf, BT_ISO_BROADCAST_CODE_SIZE); 
>  (void)memcpy(recv_state->bad_code, broadcast_code, 
>  	     sizeof(recv_state->bad_code)); 
> 
> to
> 
> ```
> 	broadcast_code = 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF;
> 	(void)memcpy(recv_state->bad_code, broadcast_code,
> 		     sizeof(recv_state->bad_code));
> ```
> 
> and
> 
> [zephyr/subsys/bluetooth/audio/bap_scan_delegator.c](https://github.com/zephyrproject-rtos/zephyr/blob/43acb93607c2161608742c59ab98aa1d77c207e5/subsys/bluetooth/audio/bap_scan_delegator.c#L952-L955)
> 
> Lines 952 to 955 in [43acb93](/zephyrproject-rtos/zephyr/commit/43acb93607c2161608742c59ab98aa1d77c207e5)
>  broadcast_code = net_buf_simple_pull_mem(buf, sizeof(internal_state->broadcast_code)); 
>   
>  (void)memcpy(internal_state->broadcast_code, broadcast_code, 
>  	     sizeof(internal_state->broadcast_code)); 
> 
> to
> 
> ```
> 	broadcast_code = 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF;
> 
> 	(void)memcpy(internal_state->broadcast_code, broadcast_code,
> 		     sizeof(internal_state->broadcast_code));
> ```

Yeah, something like that. Easier to provide review when there's a PR, but yeah, any place that we currently set the bad_code to a non-0 value, we need to set it to all-0xFF

PR #88065 (2025-04-22): bluetooth: audio: Update bad code in BASS to be v1.0.1 compliant
Fixes: #86608 
Bad Broadcast Code in BASS/Scan Delegator if BIG_Encryption field value = 0x03 (Bad_Code), Bad_Code shall be set to the value 0xFFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF according to BASS v1.0.1
closed 2025-04-22 by kartben