zephyrproject-rtos/zephyr #86651: --prep-artifacts-for-testing causes relative BOARD_DIR path

stgloorious, 2025-03-05:
Normally, `BOARD_DIR` contains a full path and is stored in `CMakeCache.txt`, but when using `--prep-artifacts-for-testing`, this path becomes relative. This fails the twister invocation if the application directory is separate from the `zephyr` directory. Specifying `--board-root` seems to have no effect.

**To Reproduce**
1. Run twister on any test suite with the `--prep-artifacts-for-testing` flag, e.g., 
~~~
west twister --prep-artifacts-for-testing -T tests/unit/crc
~~~
2. See that `BOARD_DIR` in `twister-out/unit_testing_unit_testing/host/tests/unit/crc/utilities.crc/CMakeCache.txt` is a relative path:
~~~
//Main board directory for board (unit_testing)
BOARD_DIR:PATH=subsys/testsuite/boards/unit_testing/unit_testing
~~~

**Expected behavior**
An absolute path, just when running ` west twister -T tests/unit/crc` which shows
~~~
//Main board directory for board (unit_testing)
BOARD_DIR:PATH=/home/stefan/Downloads/zephyr_test/zephyr/subsys/testsuite/boards/unit_testing/unit_testing
~~~

**Impact**
Because of this, I cannot run exported tests on a CI runner.

**Environment:**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.17.0
 - Zephyr: main (c0f76d9)

stgloorious, 2025-03-05:
To debug, I added
~~~
diff --git a/cmake/modules/boards.cmake b/cmake/modules/boards.cmake
index 3f086dca948..2ae1c0f0187 100644
--- a/cmake/modules/boards.cmake
+++ b/cmake/modules/boards.cmake
@@ -202,7 +202,10 @@ if(NOT "${ret_board}" STREQUAL "")
   set(multi_val  "DIR;REVISIONS;SOCS;QUALIFIERS")
   cmake_parse_arguments(LIST_BOARD "" "${single_val}" "${multi_val}" ${ret_board})
   list(GET LIST_BOARD_DIR 0 BOARD_DIR)
-  set(BOARD_DIR ${BOARD_DIR} CACHE PATH "Main board directory for board (${BOARD})" FORCE)
+  message(WARNING "Existing board_dir:${BOARD_DIR}")
+  set(BOARD_DIR "/foo/${BOARD_DIR}/bar" CACHE PATH "" FORCE)
+  set(BOARD_DIR "${BOARD_DIR}/foobar/" CACHE PATH "Main board directory for board (${BOARD})" FORCE)
+
   set(BOARD_DIRECTORIES ${LIST_BOARD_DIR} CACHE INTERNAL "List of board directories for board (${BOARD})" FORCE)
   foreach(dir ${BOARD_DIRECTORIES})
     set_property(DIRECTORY APPEND PROPERTY CMAKE_CONFIGURE_DEPENDS ${dir}/board.yml)
~~~
Which checks whether `BOARD_DIR` is read correctly from `LIST_BOARD_DIR` and saved to cache.

In the `build.log`, I can see that the `BUILD_DIR` CMake variable is actually the expected absolute path
~~~
CMake Warning at /home/stefan/projects/zephyr-app/zephyr/cmake/modules/boards.cmake:205 (message):
  Existing
  board_dir:/home/stefan/projects/zephyr-app/zephyr/boards/raspberrypi/rpi_pico2
Call Stack (most recent call first):
  /home/stefan/projects/zephyr-app/zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  /home/stefan/projects/zephyr-app/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/stefan/projects/zephyr-app/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

~~~
But for whatever reason, `BUILD_DIR` does not seem to be substituted properly **in the following line**, as `CMakeCache.txt` shows
~~~
//Main board directory for board (rpi_pico2)
BOARD_DIR:PATH=/fooboards/raspberrypi/rpi_pico2/bar/foobar
~~~
instead of the expected `/foo/home/stefan/projects/zephyr-app/zephyr/boards/raspberrypi/rpi_pico2/bar/foobar`.

I don't understand CMake well enough to have a clue what is going on here.

stgloorious, 2025-03-05:
Thank you @nashif for your quick response.

I see your point. This was not clear to me because this seems to be the only path where this happens. All other paths in `CMakeCache.txt` remain absolute, regardless of the `--prep-artifacts-for-testing` option.

So this means I can only unpack `twister-out` in the `zephyr` directory (or the directory that is above the appropriate `boards`), as `--board-root` does not seem to have any effect. It seems counter-intuitive to me that I cannot package the artifacts on a build machine and copy them to the same location on an equivalent test runner (if I'm running twister from my workspace). Do do have any suggestions on how to resolve this?

nashif, 2025-03-30:
probably do not use  `--prep-artifacts-for-testing` and instead package the artifacts yourself...

closed 2025-06-19 by nashif