zephyrproject-rtos/zephyr #86680: Adopt consistent format for Copyright notices

keith-zephyr, 2025-03-05:
## Introduction

The [contributor guidelines](https://docs.zephyrproject.org/latest/contribute/guidelines.html#copyrights-notices) instruct developers to follow the [Linux foundation best practices](https://www.linuxfoundation.org/blog/blog/copyright-notices-in-open-source-software-projects) for copyrights in open source projects.

### Problem description

Zephyr isn't following the Linux foundation recommendation.    Some contributors always update the copyright, many others only include a copyright message on the first commit.

### Proposed change

Adopt a standard Copyright message for Zephyr.  One of 

```
Copyright (c) YYYY The Zephyr Project Authors
Copyright (c) YYYY The Zephyr Project Contributors
Copyright (c) YYYY Zephyr Project Members
```

The first 2 options come from the Linux foundation.  The 3rd option as limited use in the tree now (15 results)

keith-zephyr, 2025-03-05:
If this is adopted, we would not remove any existing copyright notices in the tree, per the LF recommendation.

* New files would include the Zephyr Project copyright and optionally the author's own copyright.
* When modifying files, the Zephyr Project copyright is appended to the existing copyright notice for the file.

keith-zephyr, 2025-03-05:
Another discussion topic is the Copyright year format.  The Zephyr tree shows the following patterns:
```
Copyright (c) 2019
Copyright (c) 2022-2024
Copyright (c) 2017, 2019
Copyright (c) 2011-2012, 2014-2015
```

I propose we recommend only including the current year in the copyright notice.  When amending a a file, there is no need to update the date in copyright notice.  Authors are free to add a new copyright line.

For US copyright, only the year of the first publication is needed.

henrikbrixandersen, 2025-03-05:
Are you suggesting we require contributors to sign over copyright of their work to the Linux Foundation? If so, why?

nashif, 2025-03-06:
> Are you suggesting we require contributors to sign over copyright of their work to the Linux Foundation? If so, why?

where is this being suggested?

The recommendation of the LF clearly state:

```
...
These statements are intended to communicate the following:

- the work is copyrighted;
- the contributors of the code licensed it, but retain ownership of their copyrights; and
- it was licensed for distribution as part of the named project.
...
```

> ### Proposed change
> Adopt a standard Copyright message for Zephyr. One of
> 
> ```
> Copyright (c) YYYY The Zephyr Project Authors
> Copyright (c) YYYY The Zephyr Project Contributors

This one matches the recommendation, but I would recommend keeping it simple without years and (c), because this is the reason you want to go this route. notices being out of date all the time and also different companies using (c) in different ways and formats, so keep it simple and just foillow the recommendation to the letter. 

> Copyright (c) YYYY Zephyr Project Members

please no, member has a special meaning in this context, ie. being a member of the project (silver,  platinum, etc.)

> ```

keith-zephyr, 2025-03-06:
> This one matches the recommendation, but I would recommend keeping it simple without years and (c), because this is the reason you want to go this route. notices being out of date all the time and [...]

I agree  with this.  I opened this issue after seeing a PR author change the copyright year in an existing file.  If we drop the year, then it's easier for compliance checks to enforce the format as well.

keith-zephyr, 2025-03-06:
> Are you suggesting we require contributors to sign over copyright of their work to the Linux Foundation? If so, why?

As Anas noted, no one signs over copyright with this proposal.  The github history is still the authoritative source of truth for copyright.

henrikbrixandersen, 2025-03-06:
> > Are you suggesting we require contributors to sign over copyright of their work to the Linux Foundation? If so, why?
> 
> As Anas noted, no one signs over copyright with this proposal. The github history is still the authoritative source of truth for copyright.

Sounds good, sorry I misread the proposal initially. The whole "Zephyr Project Members" threw me off, as that is a distinct group of LF/Project member companies (as already pointed  by @nashif). We need our documentation to be very clear about the fact, that there is no copyright reassignment, as this is known to cause contribution issues at least for some companies.

Personally, I'd prefer the consistent copyright notice to say "The Zephyr Project Contributors" as "contributor" is already a well-established term in the Zephyr community (see e.g. https://docs.zephyrproject.org/latest/contribute/index.html).

jfischer-no, 2025-03-06:
"Copyright The XYZ Author" should not be optional, but always required when adding a new file to the project. It is established practice and does not contradict https://www.linuxfoundation.org/blog/blog/copyright-notices-in-open-source-software-projects.

butok, 2025-03-12:
> Adopt a standard Copyright message for Zephyr. One of
> 
> ```
> Copyright (c) YYYY The Zephyr Project Authors
> Copyright (c) YYYY The Zephyr Project Contributors
> Copyright (c) YYYY Zephyr Project Members
> ```
> 
> The first 2 options come from the Linux foundation. The 3rd option has limited use in the tree now (15 results)

@keith-zephyr @kartben 

The proposed change is not complete.

According to https://www.linuxfoundation.org/blog/blog/copyright-notices-in-open-source-software-projects:

"LF project communities recommend using a more general statement in a form similar to the following (where XYZ is the project’s name):
- Copyright The XYZ Authors.
- Copyright The XYZ Contributors.
- Copyright Contributors to the XYZ project.

By using a common format, the project avoids having to maintain lists of names of the authors or copyright holders, years or ranges of years, and variations on the (c) symbol."

It means:
- no year.
- no (c) sign.

aescolar, 2025-03-12:
Thanks @keith-zephyr,
I understand the motivation, but I think we need to be a carefull here[^1].

> New files would include the Zephyr Project copyright and optionally the author's own copyright.

I'm ok with recommending this, but not about enforcing it.
If a file is contributed anew, and the contributor only includes its own entity ©, that should be accepted. (Same for significant additions to existing files).

We may reject somebody adding their copyright for typo level contributions (simply because they would not be considered copyrightable). But I think that would be the limit of it.

Similarly about:

> When modifying files, the Zephyr Project copyright is appended to the existing copyright notice for the file.

If the modification is done by the same © holder, requiring this would be quite wrong. If the modification is significant, and done by a different entity, we may recommend or encourage it. But I believe it would be wrong to in any way try to enforce it.
Similarly, if a contribution is trivial, and the contributor did not add this new copyright line, blocking it would be, at best, silly.

> The github history is still the authoritative source of truth for copyright.

Until it is lost or not replicated.. Please let's be carefull about deleting/muddling copyright, authorsip, attribution or moral rights notices.

Please remember that the Apache2 and BSD licenses explicitly require to keep copyright notices.

So in short I'm ok with suggesting people to just use a format like

> Copyright (c) The Zephyr Project Contributors

when they need to add a Copyright to satisfy the copyright/license check, or when the author of a change wants to clarify that the file has been changed by somebody else than the currently listed copyright holders.

But I want to be sure that:

* Contributions will *not* be blocked on the basis of not having a "Zephyr Project Contributors" ©.
* Current © notices will not be removed (unless none of the content which was contributed under that © remains)
* That we will not, en-mass, add this kind of copyright notices to current files even when those files do not get any other change.

[^1]: When we talk about forcing somebody to add a copyright like this, or systematically adding this kind of copyright lines to current code we are crosing into something that could easily be perceived as copyright or authorship missrepresentation, or attempts to coherce others to renounce to their rights (by otherwise preventing them to contribute to the code base). We have already had a couple of blunders in the project in this area, I'd like to avoid another one.

Overall, I do not think we have a too big problem today with "too many pointless copyright lines". But I agree that it would be nice to encourage people to just add a generic "© The Zephyr Project Contributors" when that is good enough to limit the issue.

nashif, 2025-03-12:
> * New files would include the Zephyr Project copyright and optionally the author's own copyright.
> * When modifying files, the Zephyr Project copyright is appended to the existing copyright notice for the file.

We need to be careful here, 'Zephyr Project copyright' is misleading, this does not exist, you probably are referring to ' Zephyr Project Contributors' copyright. Copyrights are always with the Authors and Contributors, the Zephyr project itself does not hold the copyrights.

keith-zephyr, 2025-03-12:
> We need to be careful here, 'Zephyr Project copyright' is misleading, this does not exist, you probably are referring to ' Zephyr Project Contributors' copyright. Copyrights are always with the Authors and Contributors, the Zephyr project itself does not hold the copyrights.

Thanks for the suggestion - my initial terminology was sloppy.  My proposed edits on https://github.com/zephyrproject-rtos/zephyr/pull/86958 clarify the distinction between the "Zephyr Project" and "Zephyr Project Contributors".

closed 2025-11-11 by keith-zephyr