zephyrproject-rtos/zephyr #86681: net: lwm2m: Data cache broken by ring buffer struct changed

etiennedm, 2025-03-05:
**Describe the bug**

When enabling `CONFIG_LWM2M_RESOURCE_DATA_CACHE_SUPPORT` and the necessary options to support it, compilation fails. The root cause of this issue is a modification of `struct ring_buf` in 3075a7d9. Even though all the fields of the struct are marked as internal, the LwM2M code is using some of them to roll back the state of the ring buffer on failure.

**To Reproduce**
1. Enable the `CONFIG_LWM2M_RESOURCE_DATA_CACHE_SUPPORT` option in the `lwm2m-client` sample.
```
CONFIG_ZCBOR=y
CONFIG_ZCBOR_CANONICAL=y
CONFIG_LWM2M_RW_SENML_CBOR_SUPPORT=y
CONFIG_POSIX_TIMERS=y
CONFIG_LWM2M_RESOURCE_DATA_CACHE_SUPPORT=y
```

2. Compile.
```bash
west build -b qemu_x86 samples/net/lwm2m_client --pristine
```

**Expected behavior**
It compiles without errors.

**Logs and console output**
```
/home/etienne/dev/embedded/zephyrproject/zephyr/subsys/net/lib/lwm2m/lwm2m_message_handling.c: In function 'lwm2m_read_cached_data':
/home/etienne/dev/embedded/zephyrproject/zephyr/subsys/net/lib/lwm2m/lwm2m_message_handling.c:1406:63: error: 'struct ring_buf' has no member named 'get_base'
 1406 |                 read_info->original_get_base = cached_data->rb.get_base;
      |                                                               ^
/home/etienne/dev/embedded/zephyrproject/zephyr/subsys/net/lib/lwm2m/lwm2m_message_handling.c:1407:63: error: 'struct ring_buf' has no member named 'get_head'
 1407 |                 read_info->original_get_head = cached_data->rb.get_head;
      |                                                               ^
/home/etienne/dev/embedded/zephyrproject/zephyr/subsys/net/lib/lwm2m/lwm2m_message_handling.c:1408:63: error: 'struct ring_buf' has no member named 'get_tail'
 1408 |                 read_info->original_get_tail = cached_data->rb.get_tail;
      |                                                               ^
/home/etienne/dev/embedded/zephyrproject/zephyr/subsys/net/lib/lwm2m/lwm2m_message_handling.c: In function 'lwm2m_timeseries_data_rebuild':
/home/etienne/dev/embedded/zephyrproject/zephyr/subsys/net/lib/lwm2m/lwm2m_message_handling.c:3082:56: error: 'struct ring_buf' has no member named 'get_head'
 3082 |                 cache_temp->read_info[i].cache_data->rb.get_head =
      |                                                        ^
/home/etienne/dev/embedded/zephyrproject/zephyr/subsys/net/lib/lwm2m/lwm2m_message_handling.c:3084:56: error: 'struct ring_buf' has no member named 'get_tail'
 3084 |                 cache_temp->read_info[i].cache_data->rb.get_tail =
      |                                                        ^
/home/etienne/dev/embedded/zephyrproject/zephyr/subsys/net/lib/lwm2m/lwm2m_message_handling.c:3086:56: error: 'struct ring_buf' has no member named 'get_base'
 3086 |                 cache_temp->read_info[i].cache_data->rb.get_base =
      |                                                        ^
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.3
 - Commit SHA: 0717698277c

nashif, 2025-03-06:
@npitre can you please take a look?

fabiobaltieri, 2025-03-06:
tagging as blocker since it's a regression introduced in the 4.1 cycle

PR #86683 (2025-03-06): net: lwm2m: Fix data cache rollback logic compilation
The root cause of this issue is a modification of `struct ring_buf` in 3075a7d9. Even though all the fields of the struct are marked as internal, the LwM2M code is using some of them to roll back the state of the ring buffer on failure.

Fixes #86681.
closed 2025-03-06 by fabiobaltieri