zephyrproject-rtos/zephyr #86684: doc: Documentation issue in 'develop/getting_started/index' - not installing GIT

asbrack, 2025-03-05:
**Describe the bug**

When installing the latest Zephyr on a computer that does not already have a copy of 'git' in the search path, the installation indicates that 'git' has been successfully installed, but it cannot be found in the search path. All other copies of 'git' on my system are local to the various applications: nRF, VScode, ? (simplified output from debugging steps)

> PS C:\Windows\system32> choco install git --force
> Chocolatey v2.4.3
> Installing the following packages:
> git
> By installing, you accept licenses for the packages.
> git v2.48.1 already installed. Forcing reinstall of version '2.48.1'.
>  Please use upgrade if you meant to upgrade to a new version.
> Downloading package from source 'https://community.chocolatey.org/api/v2/'
> Progress: Downloading git 2.48.1... 100%
> 
> git v2.48.1 (forced) [Approved]
> git package files install completed. Performing other installation steps.
>  The install of git was successful.
>   Deployed to 'C:\ProgramData\chocolatey\lib\git'
> 
> Chocolatey installed 1/1 packages.
>  See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

 
> PS C:\Windows\system32> git
> git : The term 'git' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the
> spelling of the name, or if a path was included, verify that the path is correct and try again.
> At line:1 char:1
> + git
> + ~~~
>     + CategoryInfo          : ObjectNotFound: (git:String) [], CommandNotFoundException
>     + FullyQualifiedErrorId : CommandNotFoundException

When the command "west init zephyrproject" is executed, you get an error message indicating that 'git' cannot be found. My resolution was to install a global copy from the git distribution site and manually add it to the global PATH environment variable. After that I was able to complete the "Getting Started" page.

C:\Users\tony\Documents\work>dir C:\ProgramData\chocolatey\bin
 Volume in drive C is Windows
 Volume Serial Number is 8C18-B231

 Directory of C:\ProgramData\chocolatey\bin

2025-03-05  01:46 PM    <DIR>          .
2025-03-05  01:46 PM    <DIR>          ..
2025-03-03  04:28 PM           391,680 7z.exe
2025-02-27  02:04 PM           146,984 choco.exe
2025-03-05  01:46 PM           392,704 convert-dtsv0.exe
2025-03-05  01:46 PM           392,704 cygpath.exe
2025-03-05  01:46 PM           392,704 dtc.exe
2025-03-05  01:46 PM           392,704 fdtdump.exe
2025-03-05  01:46 PM           392,704 fdtget.exe
2025-03-05  01:46 PM           392,704 fdtoverlay.exe
2025-03-05  01:46 PM           392,704 fdtput.exe
2025-03-05  01:46 PM           392,704 getprocaddr32.exe
2025-03-05  01:46 PM           392,704 getprocaddr64.exe
2025-03-05  01:42 PM           392,704 gperf.exe
2025-03-05  01:42 PM           392,704 ninja.exe
2025-03-05  01:46 PM            99,328 python3.11.exe
2025-02-27  02:04 PM             4,495 RefreshEnv.cmd
2025-03-05  01:46 PM           392,704 wget.exe
2025-03-03  05:57 PM           392,704 zstd.exe
2025-03-03  01:32 PM                21 _processed.txt
              18 File(s)      5,747,660 bytes
               2 Dir(s)  820,053,069,824 bytes free

C:\Users\tony\Documents\work>dir C:\ProgramData\chocolatey\lib\git
 Volume in drive C is Windows
 Volume Serial Number is 8C18-B231

 Directory of C:\ProgramData\chocolatey\lib\git

2025-03-05  01:48 PM    <DIR>          .
2025-03-05  01:48 PM    <DIR>          ..
2025-03-05  01:48 PM             3,772 git.nupkg
2025-02-13  01:12 PM             3,291 git.nuspec
               2 File(s)          7,063 bytes
               2 Dir(s)  820,052,340,736 bytes free

C:\Users\tony\Documents\work>

**Environment**

* Page: `develop/getting_started/index`
* Version: main
* SHA-1: c0c6047257663bbb21d98c75f4da13aceecc5b98

kartben, 2025-03-06:
@asbrack can you confirm you ran chocolatey from a terminal launched "as Administrator"?

asbrack, 2025-03-07:
It was run from a PowerShell window (not a terminal) launched as "administrator" and the actual installation command was run multiple times. All of the other commands were installed (see directory listing). As a debug step "choco install git --force" was run again. See the directory listing that I provided.

I can still repeat this at will, although I eventually installed a standalone copy of 'git' from their distribution site to get around the issue.

git v2.48.1 (forced) [Approved]
git package files install completed. Performing other installation steps.
The install of git was successful.
Deployed to 'C:\ProgramData\chocolatey\lib\git'
- - -
Directory of C:\ProgramData\chocolatey\lib\git

2025-03-05 01:48 PM

.
2025-03-05 01:48 PM ..
2025-03-05 01:48 PM 3,772 git.nupkg
2025-02-13 01:12 PM 3,291 git.nuspec

closed 2025-08-12 by github-actions[bot]