zephyrproject-rtos/zephyr #86689: NXP LPSPI driver not configuring correct chip select

decsny, 2025-03-05:
**Describe the bug**

The LPSPI driver is 
- hardcoding to always slave select 0
- not respecting the active high flag from the API

This hardcoding is regression from 4.0

**To Reproduce**
I do not know of any in tree sample to reproduce with, but the observation about this in the code and the fix is relatively trivial.

**Expected behavior**
Driver respects the spi_config passed in the API.

**Impact**
Since multi slave use case is not tested in Zephyr yet, from my perspective I have no impact, but for some users, maybe it has an impact if they used a slave select other than 0 with active low, but I don't have information about that.

**Logs and console output**
N/A

**Environment (please complete the following information):**
Zephyr 4.1.0-rc3

PetervdPerk-NXP, 2025-03-06:
Bosch BMI088 has 2 chip selects which is on the NXP VMU_RT1170 board.
https://github.com/zephyrproject-rtos/zephyr/blob/3d0a7dab1a975dddb3624f46006d301896475469/boards/nxp/vmu_rt1170/vmu_rt1170_mimxrt1176_cm7.dts#L303-L305
For now the workaround was to use GPIO chip select.
But this bug report indicates why hw chip select was not working

ofirshe, 2025-03-06:
Hey @PetervdPerk-NXP, 

Do you know anyone at NXP with a board that has two or more SPI slaves on an LPSPI peripheral using hardware chip select, who could help verify that the [fix](https://github.com/zephyrproject-rtos/zephyr/pull/86687) resolves the issue?

closed 2025-03-06 by fabiobaltieri