zephyrproject-rtos/zephyr #86697: ERROR:

Rohanhc2001, 2025-03-06:
SOC_MIMXRT1062 (defined at soc/nxp/imxrt/imxrt10xx/Kconfig.soc:35,
soc/nxp/imxrt/imxrt10xx/Kconfig:91) is assigned in a configuration file, but is not directly user-
configurable (has no prompt). It gets its value indirectly from other symbols. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SOC_MIMXRT1062 and/or look up
SOC_MIMXRT1062 in the menuconfig/guiconfig interface. The Application Development Primer, Setting

henrikbrixandersen, 2025-03-06:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Also, please use a more descriptive title when opening an issue.

decsny, 2025-03-06:
please find a support channel such as discord or nxp community for help with defining custom board in zephyr, if this is actually a bug report then please fill out the template and reopen with more information