zephyrproject-rtos/zephyr #86698: SPI transmission and reception are not functioning as expected for stm32g0c1ret3 controller

RiyaKori-eaton, 2025-03-06:
**Describe the bug**
 NA

**To Reproduce**
1.  The pin configurations which was supported by our microcontroller has been done .
2. Necessary files have been updated, including the DTS file, overlay file, prj.conf file, and CMakeLists.txt.
3. The build process has been completed, and the firmware has been successfully flashed onto the microcontroller.

**Expected behavior**
  In SPI (Serial Peripheral Interface) communication, if values are transmitted via the transmission buffer, corresponding values should be received in the receiver buffer. This is a fundamental aspect of SPI's full-duplex communication capability.

**Impact**
 showstopper

**Logs and console output**

**In Zephyr 3.6.0**
we checked SPI clock on nucleo-g071rb and nucleo-g474re
In  nucleo-g071rb we are not able to see sclk waveform and nucleo-g474re series its working fine
 
below screenshot is for nucleo-g474re

[image]

**In Zephyr 4.0.0**
we checked SPI clock on nucleo-g071rb and nucleo-g474re
In  nucleo-g071rb we are not able to see sclk waveform and nucleo-g474re series its working fine
For nucleo-g071rb we observed same issue there is no activity on sclk

[image]

for nucleo-g474re sclk we are able to see the waveform

[image]

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)  
 - Windows
 - Toolchain (e.g Zephyr SDK, ...)
 - Zephyr 3.6.0
 - Commit SHA or Version used
 
[image]

[image]
[image]
[image]

 - NA

**Additional context**
[](url) @AmolMeshram-eaton
@bhardadinesh-eaton

realhonbo, 2025-03-18:
I thought it maybe a old bug of hardware nss caused, it occurred in stm32f1xx and stm32f4xx, and fixed in stm32h7xx:
pb-0 is the hardware nss of alternate spi1, Even though you choose it to soft-nss, may dtc processed it to hard-nss? Could you have a try to change a cs pin?

[image]

RiyaKori-eaton, 2025-06-06:
We have also attempted this on Zephyr v4.0.0, but the clock signal is still not observed. Please refer to the attached screenshots for further details @JarmouniA

gautierg-st, 2025-07-01:
Hi @RiyaKori-eaton,
I've tried to reproduce your issue with a Nucleo G071RB and a Nucleo G0B1RE, in v3.7.0, v4.0.0 and current main, with default SPI pins and your own, with CS as GPIO or as HW NSS, but in all cases, I see all signals and tests/drivers/spi/spi_loopback is successful.

Do you have a particular setup that could explain your issue?