zephyrproject-rtos/zephyr #86718: USB NEXT: Error while adding interface string descriptor

marwaiehm-st, 2025-03-06:
**Describe the bug**

We are experiencing an issue with the samples/subsys/usb/cdc_acm sample in Zephyr. The sample works, but we encounter an error while adding the interface string descriptor for the USB CDC ACM. The error message is as follows:
```shell
*** Booting Zephyr OS build v4.1.0-rc3-23-g14f382b3b61c ***
[00:00:00.010,000] <err> usbd_cdc_acm: Failed to add interface string descriptor
[00:00:00.010,000] <inf> cdc_acm_echo: USB device support enabled
[00:00:00.010,000] <inf> cdc_acm_echo: Wait for DTR
[00:00:00.015,000] <inf> cdc_acm_echo: USBD message: Device suspended
[00:00:00.185,000] <inf> cdc_acm_echo: USBD message: Bus reset
[00:00:00.299,000] <inf> cdc_acm_echo: USBD message: Bus reset
[00:00:00.400,000] <inf> cdc_acm_echo: USBD message: New device configuration
[00:00:00.402,000] <inf> cdc_acm_echo: USBD message: CDC ACM line coding
[00:00:00.402,000] <inf> cdc_acm_echo: Baudrate 9600
[00:00:00.946,000] <inf> cdc_acm_echo: USBD message: CDC ACM control line state
[00:00:00.946,000] <inf> cdc_acm_echo: DTR set
[00:00:00.946,000] <inf> cdc_acm_echo: USBD message: CDC ACM line coding
[00:00:00.946,000] <inf> cdc_acm_echo: Baudrate 115200

``` 

- The issue appears only with boards that have USB high-speed (USB_HS) support.
- We reproduced the issue on the boards:  stm32h747i_disco, stm32f769i_disco and nucleo_u5a5zj_q

**To Reproduce**

1.  west build -p always -b stm32h747i_disco/stm32h747xx/m7  samples/subsys/usb/console/ -- -DCONF_FILE=usbd_next_prj.conf
2. west flash

**Expected behavior**

 The sample should run without any errors, and the interface string descriptor should be added successfully.

**Environment (please complete the following information):**

 - Zephyr version: v4.0.0-4508-g2ba4aeaa0490

**Additional context**

jfischer-no, 2025-03-06:
>The sample should run without any errors, and the interface string descriptor should be added successfully.

I am pretty sure it does. `<err> usbd_cdc_acm: Failed to add interface string descriptor` may be annoying, but it is not a problem. Also, your report is incorrect, only samples/subsys/usb/cdc-acm has label property.

marwaiehm-st, 2025-03-06:
> I am pretty sure it does. `<err> usbd_cdc_acm: Failed to add interface string descriptor` may be annoying, but it is not a problem. Also, your report is incorrect, only samples/subsys/usb/cdc-acm [...]

The issue is not blocking, but it would be beneficial to address it for future users.

carlescufi, 2025-04-01:
@jfischer-no @marwaiehm-st should we close this issue then?

PR #88004 (2025-04-02): usb: device_next: avoid false error logging in CDC ACM
The interface descriptor and its associated string descriptor are shared between different speed configurations. Do not try to add a string descriptor if it has already been added and the index is not zero.

Fixes: #86718
closed 2025-04-02 by kartben