zephyrproject-rtos/zephyr #86720: Feature request: Generation of release artifacts

MarkoSagadin, 2025-03-06:
**Is your feature request related to a problem? Please describe.**

In our firmware development workflow, release artifact generation is an essential process that ensures built binaries and related files are organized, named, and packaged in a way that makes them easily accessible and understandable to end users. This structure simplifies firmware distribution and deployment, particularly for users who are not familiar with Zephyr's build system.

By following a structured artifact generation process, we ensure:

- Clear versioning and traceability of firmware builds.
- Easy navigation for users downloading firmware from GitHub releases.
- Retention of necessary files while removing unnecessary ones.

Our current release folder structure looks like this:

```
release
├── apps
│   └── app
│       ├── debug
│       │   ├── custom_nrf52840dk
│       │   │   ├── app-custom_nrf52840dk-v1.0.0-debug-99d4fc7.bin
│       │   │   ├── app-custom_nrf52840dk-v1.0.0-debug-99d4fc7.elf
│       │   │   └── app-custom_nrf52840dk-v1.0.0-debug-99d4fc7.hex
│       │   └── nrf52840dk_nrf52840
│       │       ├── app-nrf52840dk_nrf52840-v1.0.0-debug-99d4fc7.bin
│       │       ├── app-nrf52840dk_nrf52840-v1.0.0-debug-99d4fc7.elf
│       │       └── app-nrf52840dk_nrf52840-v1.0.0-debug-99d4fc7.hex
│       ├── release
│       │   ├── custom_nrf52840dk
│       │   │   ├── app-custom_nrf52840dk-v1.0.0-99d4fc7.bin
│       │   │   ├── app-custom_nrf52840dk-v1.0.0-99d4fc7.elf
│       │   │   └── app-custom_nrf52840dk-v1.0.0-99d4fc7.hex
│       │   └── nrf52840dk_nrf52840
│       │       ├── app-nrf52840dk_nrf52840-v1.0.0-99d4fc7.bin
│       │       ├── app-nrf52840dk_nrf52840-v1.0.0-99d4fc7.elf
│       │       └── app-nrf52840dk_nrf52840-v1.0.0-99d4fc7.hex
│       └── uart
│           ├── custom_nrf52840dk
│           │   ├── app-custom_nrf52840dk-v1.0.0-uart-99d4fc7.bin
│           │   ├── app-custom_nrf52840dk-v1.0.0-uart-99d4fc7.elf
│           │   └── app-custom_nrf52840dk-v1.0.0-uart-99d4fc7.hex
│           └── nrf52840dk_nrf52840
│               ├── app-nrf52840dk_nrf52840-v1.0.0-uart-99d4fc7.bin
│               ├── app-nrf52840dk_nrf52840-v1.0.0-uart-99d4fc7.elf
│               └── app-nrf52840dk_nrf52840-v1.0.0-uart-99d4fc7.hex
├── app-v1.0.0-99d4fc7.zip
├── app-v1.0.0-debug-99d4fc7.zip
└── app-v1.0.0-uart-99d4fc7.zip
```

Until now, this process has been managed by our in-house tool, East, which automates firmware builds, final structuring and renaming. However, we are now migrating to Zephyr’s Twister framework as it provides greater flexibility and aligns better with the Zephyr ecosystem. The issue we face is that Twister can generate build artifacts (more specifically it creates build folders) but lacks functionality for renaming and packaging them into a structured format. Additionally, it would be a quite hard task for a non-Zephyr user to navigate the build directory and pick correct file which they would have to flash/DFU to the target.

Twister's `--prep-artifacts-for-testing` flag partially addresses this but does not fully meet our needs. The resulting folder structure includes unnecessary files and does not retain Nordic's build system-generated files (like dfu_application.zip).

**Describe the solution you'd like**

We propose enhancing Twister with an optional feature to rename and package release artifacts into a structured format. This could be implemented in one of the following ways:

As an additional Twister flag to enable structured artifact packaging.

As a separate west command that processes the twister-out folder and generates the final release directory.

In both cases the there will probably be a need for some file that specifies the naming convention of how the files are named and which files to rename. Although the tool could offer some sane defaults, this pretty much use-case specific.

I should add that although the above example only lists app firmware, the tool should also work with samples or tests, in other words they wouldn't be treated any differently.

Describe alternatives you've considered

Use Twister as is, and add only renaming part to our internal tool.

**Additional context**

Our current approach using East follows a structured artifact naming and packaging convention. More details on our process can be found here:

- [East release command documentation](https://github.com/IRNAS/irnas-east-software/blob/main/docs/configuration.md#release-command)
- [Zephyr Developer Summit talk on East](https://www.youtube.com/watch?v=MaHuHvuN72Y)
- [Current naming convention guidelines](https://github.com/IRNAS/irnas-guidelines-docs/blob/main/docs/github_projects_guidelines.md#release-artifacts-naming-scheme-)

Keep in mind: our artifact naming convention works for us (or it at least did, until we started using XIP in our projects), but it probably won't work for the this possible enhancement.Do members of Zephyr community have the similar artifact packaging needs? If yes, how are they currently resolved?

Would the Zephyr community find value in such an enhancement? If so, which approach would be preferable for upstream contribution?

henrikbrixandersen, 2025-03-06:
Twister is intended to be a test runner, not a release building tool.

MarkoSagadin, 2025-03-13:
That is correct, I mentioned the Twister as there is a significant overlap between what Twister does and what is needed for the release artifact generation process.

As I wrote, a separate west command is also an option.