zephyrproject-rtos/zephyr #86799: DTR empty timeout error while running Ztest for Xilinx AXI Quad SPI driver

AmitMahapatra, 2025-03-08:
**Describe the bug**
While running Ztest in non-async mode for the Xilinx AXI Quad SPI driver on an mbv32 platform with Micron mt25qu01g SPI-NOR flash connected onboard, we observed a DTR Empty TIMEOUT error, leading to the failure of the "test_read_unaligned_address" test.

Upon performing a git bisect, we identified https://github.com/zephyrproject-rtos/zephyr/commit/680fa154bf7896e2a0aa8d973d0422097c6d13aa commit as the root cause of the issue

**To Reproduce**
- Use a device tree with a axi-quadspi node and a flash node.
- Build tests/drivers/flash/common/ and run

`axi_spi0: spi@44a00000 {
                status = "okay";
                #address-cells = <0x01>;
                #size-cells = <0x00>;
                fifo-size = <256>;
                reg = <0x44a00000 0x10000>;
                compatible = "xlnx,xps-spi-2.00.a";
                interrupt-parent = <&axi_intc>;
                xlnx,startup-block;
                interrupts = < 3 0 >;
                xlnx,num-ss-bits = <0x2>;
                xlnx,num-transfer-bits = <8>;
                flash@0 {
                        compatible = "jedec,spi-nor";
                        reg = <0x00>;
                        spi-max-frequency = <8000000>;
                        jedec-id = [20 bb 21]; /* mt25qu01g */
                        size = <134217728>;
                        has-dpd;
                        t-enter-dpd = <10000>;
                        t-exit-dpd = <35000>;
                };
}
`

**Expected behavior**
Running TESTSUITE flash_driver

Test will run on device flash@0
START - test_flash_fill
 PASS - test_flash_fill in 41.134 seconds

START - test_flash_flatten
 PASS - test_flash_flatten in 41.140 seconds

START - test_read_unaligned_address
 PASS - test_read_unaligned_address in 0.241 seconds

TESTSUITE flash_driver succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [flash_driver]: pass = 3, fail = 0, skip = 0, total = 3 duration = 82.515 seconds
 - PASS - [flash_driver.test_flash_fill] duration = 41.134 seconds
 - PASS - [flash_driver.test_flash_flatten] duration = 41.140 seconds
 - PASS - [flash_driver.test_read_unaligned_address] duration = 0.241 seconds

------ TESTSUITE SUMMARY END ------

PROJECT EXECUTION SUCCESSFUL

**Impact**
While sending small bust of data this issue can be reproduced. 

**Logs and console output**

Running TESTSUITE flash_driver

Test will run on device flash@0
START - test_flash_fill
PASS - test_flash_fill in 41.134 seconds

START - test_flash_flatten
PASS - test_flash_flatten in 41.140 seconds

START - test_read_unaligned_address
E: DTR empty timeout
 
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:165: flash_driver_test_read_unaligned_address: (rc not equal to 0)
Cannot read flash
FAIL - test_read_unaligned_address in 0.483 seconds

TESTSUITE flash_driver failed.
 
------ TESTSUITE SUMMARY START ------
 
SUITE FAIL -  66.67% [flash_driver]: pass = 2, fail = 1, skip = 0, total = 3 duration = 82.757 seconds
- PASS - [flash_driver.test_flash_fill] duration = 41.134 seconds
- PASS - [flash_driver.test_flash_flatten] duration = 41.140 seconds
- FAIL - [flash_driver.test_read_unaligned_address] duration = 0.483 seconds
 
------ TESTSUITE SUMMARY END ------
 
PROJECT EXECUTION SUCCESSFUL

**Environment (please complete the following information):**

 - Zephyr v3.7.0

**Additional context**

robhancocksed, 2025-03-10:
@AmitMahapatra Can you provide more details on the failure? The log posted above appears to show a successful test run.

AmitMahapatra, 2025-03-11:
Hello @robhancocksed 

Sorry, my bad I accidentally posted the expected output in "Logs and Console Output". I have now updated the logs with the actual failure logs.

Regards,
Amit

robhancocksed, 2025-03-11:
@AmitMahapatra The patch in question introduced a timeout on SPI commands, previously I don't think transfers would ever time out. The timeout is set to 20 ms plus the value of CONFIG_SPI_COMPLETION_TIMEOUT_TOLERANCE. Can you try increasing that option to a much higher value initially for the test and see if the timeout is still occurring? Maybe something is just happening slower than expected..

AmitMahapatra, 2025-03-12:
Hello @robhancocksed 

In  [xlnx_quadspi_start_tx( )](https://github.com/zephyrproject-rtos/zephyr/blob/c4ebf713b10ce63f44ee3497159c54377ecc839b/drivers/spi/spi_xlnx_axi_quadspi.c#L233), when the driver toggles the inhibit bit after filling the TX FIFO, the data in the TX FIFO is immediately transmitted onto the SPI bus. While operating in non-async mode for small data bursts (less than 2 bytes), there is a possibility that the DTR Empty interrupt is triggered before the calling thread starts waiting for it. Consequently, the calling thread never receives the event, leading to a transfer failure with a DTR empty timeout error.

I increased the timeout to as much as 10 seconds, but the issue still persists.

Regards,
Amit

robhancocksed, 2025-03-12:
@AmitMahapatra Looking at the code, I am not sure how that scenario would be happening, since the dtr_empty event is cleared prior to xlnx_quadspi_start_tx being called. If the interrupt has already occurred and posted the event by the time we wait for completion, then since the k_event_wait call is not specifying to clear the event prior to waiting, that call should return immediately in that case. Are you sure that is the sequence of events that is happening?

AmitMahapatra, 2025-03-17:
Hello @robhancocksed,

I believe that when the driver attempts to post the dtr_empty event, it is not reaching the waiting thread, causing the thread to time out. As an experiment, without reverting the patch, I replaced the  [k_event_post( )](https://github.com/zephyrproject-rtos/zephyr/blob/7431e52f1dc44df50052cb5eb8bedd1ea16f3e55/drivers/spi/spi_xlnx_axi_quadspi.c#L498) call with k_event_set(). With this change, I ran the Ztest multiple times, and the DTR timeout issue did not occur—the Ztest completed successfully every time. I am looking into the implementation of both APIs to understand the differences in their behavior in this scenario.

Regards,
Amit

robhancocksed, 2025-03-17:
@AmitMahapatra The only difference between k_event_post and k_event_set is that the latter replaces the pending events with the provided mask rather than adding to it. In the case of this driver where only a single bit is used and k_event_post is only ever called passing 1, I don't see how this could cause a behavior change. Are you sure that difference is reproducible?

AmitMahapatra, 2025-03-27:
Hello @robhancocksed 

I ran the Ztest multiple times and was able to consistently reproduce the issue with k_event_post. However, the issue does not occur when using k_event_set.

Regards,
Amit

robhancocksed, 2025-03-27:
@AmitMahapatra Could you add some debugging code to the driver to print out the events value inside the k_event structure before and after the k_event_post call and see if that is different when using k_event_set? I don't really know how to explain why those are behaving differently in this case.

closed 2025-08-12 by github-actions[bot]