zephyrproject-rtos/zephyr #86804: Add hal_xhsc repo to support XHSC SoCs

TrackingRobot, 2025-03-08:
## Origin
Q: 
Name of project hosting the original open source code:
A: 
[zephyr_hc32_hal]

Q: 
Provide a link to the source:
A: 
https://github.com/levizh/zephyr_hc32_hal.git

## Purpose
Q: 
Brief description of what this software does:
A: 
This software provides the Device Driver Layer (DDL) for Xiaohua Semiconductor's (XHSC) HC32F series SoCs. 

## Mode of integration
Q: 
Describe whether you'd like to integrate this external component in the main tree or as a module, and why:
A: 
As with other manufacturers' SoCs, we recommend creating a repo (hal_xhsc) that treats the DDL component as a separate module. This approach allows for better modularity and maintainability, as it isolates the hardware-specific code from the main Zephyr tree. It also makes it easier to update and maintain the HAL without affecting the core Zephyr code base.

## Maintainership
Q: 
List the person(s) that will be maintaining the integration of this external code for the foreseeable future:
A: 
@Levi Zhang(zhangxl@xhsc.com.cn), maintainer, employee of XHSC
@Biao Peng(pengbiao@xhsc.com.cn), employee of XHSC
@Chuang Liang(liangchuang@xhsc.com.cn), employee of XHSC
@Lixin Zhang(zhanglx@xhsc.com.cn), employee of XHSC
@JinPing Yang(Yangjp@xhsc.com.cn), employee of XHSC
@Yongbing Liu(liuyb@xhsc.com.cn, code9527@126.com), employee of XHSC

## Pull Request
Q: 
Pull request (if any) with the actual implementation of the integration:
A: 
We will submit zephyr PR after our DDL repository (hal_xhsc) is created.

## Description

Q: 
What is its primary functionality (e.g., SQLLite is a lightweight
database)?
A: 
Provides SoC modules' interfaces to directly manipulate XHSC SoCs hardware module registers. 
For example, ADC_Init(), ADC_DeInit(), ...

Q: 
What problem are you trying to solve? (e.g., a state store is
required to maintain ...)
A: 
To provide hardware driver support for XHSC SoCs running Zephyr.

Q: 
Why is this the right component to solve it (e.g., SQLite is small,
easy to use, and has a very liberal license.)
A: 
This is the official driver library released to optimally match XHSC SoCs.

## Dependencies

Q: 
What other components does this package depend on?
A: 
XHSC SoC DDL is developed independently and does not depend on other package.

Q: 
Will the Zephyr project have a direct dependency on the component, or
will it be included via an abstraction layer with this component as a
replaceable implementation?
A: 
The Zephyr project based on XHSC SoCs will have a direct dependency on this component. However, the DDL itself is designed to be modular, allowing for easy replacement or extension of specific hardware drivers if needed.

## Revision
Q: 
Version or SHA you would like to integrate initially
A: 
We want to integrate the latest commit(fc37871336e48b57829a0fb529cd6885cffd6da5) from the repository(https://github.com/levizh/zephyr_hc32_hal.git).

## License
Q: 
Please use an SPDX identifier (https://spdx.org/licenses/), such as
``BSD-3-Clause``.
A: 
Yes, our DDL is all open source and the source files almost always have SPDX.

jfischer-no, 2025-03-14:
It looks like another STM32F4 family clone to me.
What is the name and website of the manufacturer?
Where can we find the datasheet in English?

TrackingRobot, 2025-03-15:
Thanks for your reply.  You are great to remind me to paste the important information about our company, the vendor of the HC32F4A0.

> It looks like another STM32F4 family clone to me.  
It is based on the Cortex-M4, similar to the STM32F4 series. However, it has significant differences from the STM32F4 family in terms of memory, flash, and some other peripherals. Actually, our HC32F series MCUs are pin-to-pin compatible with STM32F4 chips, but they are not clones. In my opinion, MCUs that clone the STM32F4 should be able to use STM32F4 firmware. The MCU naming is similar to ST just to accommodate STM32F4 usage. 

>What is the name and website of the manufacturer? 
The manufacturer is Xiaohua Semiconductor Company Limited (XHSC), which is a tertiary company of China Electronics and Information Industry Corporation (CEC). For more information about our company, please click https://www.xhsc.com.cn. Note that the website is in Chinese.  : (

>Where can we find the datasheet in English?
The English version of the datasheet is available at [DS_HC32F4A0SeriesDatasheet_Rev1.4](https://oss-nc-beijing-2.cecloudcs.com/doc-ds/DS_HC32F4A0SeriesDatasheet_Rev1.4.pdf)

Pengbiao12345, 2025-03-20:
> ## Origin
> Q: Name of project hosting the original open source code: A: [zephyr_hc32_hal]
> 
> Q: Provide a link to the source: A: https://github.com/levizh/zephyr_hc32_hal.git
> 
> ## Purpose
> Q: Brief description of what this software does: A: This software provides the Device Driver Layer (DDL) for Xiaohua Semiconductor's (XHSC) HC32F series SoCs.
> 
> ## Mode of integration
> Q: Describe whether you'd like to integrate this external component in the main tree or as a module, and why: A: As with other manufacturers' SoCs, we recommend creating a repo (hal_xhsc) that treats the DDL component as a separate module. This approach allows for better modularity and maintainability, as it isolates the hardware-specific code from the main Zephyr tree. It also makes it easier to update and maintain the HAL without affecting the core Zephyr code base.
> 
> ## Maintainership
> Q: List the person(s) that will be maintaining the integration of this external code for the foreseeable future: A: [@levi](https://github.com/levi) Zhang([zhangxl@xhsc.com.cn](mailto:zhangxl@xhsc.com.cn)), maintainer, employee of XHSC [@Biao](https://github.com/Biao) Peng([pengbiao@xhsc.com.cn](mailto:pengbiao@xhsc.com.cn)), employee of XHSC [@chuan](https://github.com/chuan) Liang([liangchuan@xhsc.com.cn](mailto:liangchuan@xhsc.com.cn)), employee of XHSC [@lixin](https://github.com/lixin) Zhang([zhanglx@xhsc.com.cn](mailto:zhanglx@xhsc.com.cn)), employee of XHSC [@jinping](https://github.com/jinping) Yang([yangjp@xhsc.com.cn](mailto:yangjp@xhsc.com.cn)), employee of XHSC [@yongbing](https://github.com/yongbing) Liu([liuyb@xhsc.com.cn](mailto:liuyb@xhsc.com.cn), [code9527@126.com](mailto:code9527@126.com)), employee of XHSC
> 
> ## Pull Request
> Q: Pull request (if any) with the actual implementation of the integration: A: We will submit zephyr PR after our DDL repository (hal_xhsc) is created.
> 
> ## Description
> Q: What is its primary functionality (e.g., SQLLite is a lightweight database)? A: Provides SoC modules' interfaces to directly manipulate XHSC SoCs hardware module registers. For example, ADC_Init(), ADC_DeInit(), ...
> 
> Q: What problem are you trying to solve? (e.g., a state store is required to maintain ...) A: To provide hardware driver support for XHSC SoCs running Zephyr.
> 
> Q: Why is this the right component to solve it (e.g., SQLite is small, easy to use, and has a very liberal license.) A: This is the official driver library released to optimally match XHSC SoCs.
> 
> ## Dependencies
> Q: What other components does this package depend on? A: XHSC SoC DDL is developed independently and does not depend on other package.
> 
> Q: Will the Zephyr project have a direct dependency on the component, or will it be included via an abstraction layer with this component as a replaceable implementation? A: The Zephyr project based on XHSC SoCs will have a direct dependency on this component. However, the DDL itself is designed to be modular, allowing for easy replacement or extension of specific hardware drivers if needed.
> 
> ## Revision
> Q: Version or SHA you would like to integrate initially A: We want to integrate the latest commit(fc37871336e48b57829a0fb529cd6885cffd6da5) from the repository(https://github.com/levizh/zephyr_hc32_hal.git).
> 
> ## License
> Q: Please use an SPDX identifier (https://spdx.org/licenses/), such as `BSD-3-Clause`. A: Yes, our DDL is all open source and the source files almost always have SPDX.

Hello zephyr teams.
About the Revsion. we provided the "SOC_RESET_HOOK" feature, so we desirable that the latest commit(fc51c1a31ede401b5b5999b48b6c4fe98c10f5f2) is integrated.

TrackingRobot, 2025-03-20:
Do we need to make any changes to the contents of the device driver library?
Could someone please assist us with this issue?
Thank you very much.

TrackingRobot, 2025-03-21:
> Hey [@TrackingRobot](https://github.com/TrackingRobot), this has been introduced to the TSC at yesterday's meeting, will get discussed again in a week, expect an update by then, so far I don't think no one suggested or requested any changes, but there should be more people from the project looking at it these days.

Hi @fabiobaltieri, thanks for your response. 
We will continue to monitor this issue. If TSC has any suggestions, we will respond as soon as possible.

TrackingRobot, 2025-04-14:
Comment to keep this issue active.

TrackingRobot, 2025-07-25:
We have prepared the PR branch https://github.com/levizh/zephyr_hc32_hal/tree/hal_xhsc based on the latest commit. Please help us create zephyrproject-rtos/hal_xhsc at your convenience. Thank you very much.

carlescufi, 2025-07-30:
> We have prepared the PR branch https://github.com/levizh/zephyr_hc32_hal/tree/hal_xhsc based on the latest commit. Please help us create zephyrproject-rtos/hal_xhsc at your convenience. Thank you very much.

@stephanosio could you help with that please?

stephanosio, 2025-08-01:
https://github.com/zephyrproject-rtos/hal_xhsc has been created.

Please add a corresponding entry in the MAINTAINERS file with the list of maintainers and collaborators for write access to the module repository.

TrackingRobot, 2025-08-01:
> https://github.com/zephyrproject-rtos/hal_xhsc has been created.
> 
> Please add a corresponding entry in the MAINTAINERS file with the list of maintainers and collaborators for write access to the module repository.

@stephanosio Thank you very much. We are glad to see the repo has been created. We will complete the MAINTAINERS file as part of our PR.

TrackingRobot, 2025-08-22:
Hi @stephanosio, may I ask if we need to submit a PR to the MAINTAINERS file first to get access to the hal_xhsc module? I have made a PR [#94044]([https://github.com/zephyrproject-rtos/zephyr/pull/94044]) that consists of multiple commits. However, the PR failed some CI tests due to a lack of write access to the hal_xhsc module.

TrackingRobot, 2025-08-25:
> Hi [@stephanosio](https://github.com/stephanosio), may I ask if we need to submit a PR to the MAINTAINERS file first to get access to the hal_xhsc module? I have made a PR [...]

Submitted a PR [MAINTAINERS PR](https://github.com/zephyrproject-rtos/zephyr/pull/94941) to obtain hal_xhsc write access.

closed 2025-09-15 by TrackingRobot