zephyrproject-rtos/zephyr #86817: Help text BT_PERIPHERAL_PREF_MAX_INT wrong?

robertinant, 2025-03-09:
https://github.com/zephyrproject-rtos/zephyr/blob/df5318479b914d320f52870c6dfd0420779fc5c0/subsys/bluetooth/host/Kconfig.gatt#L238
`range` indicates 6 but help says 3200. Which is it? Or am I missing something?

```
config BT_PERIPHERAL_PREF_MAX_INT
	int "Peripheral preferred maximum connection interval in 1.25ms units"
	default 40
	range 6 $(UINT16_MAX) if !BT_CONN_PARAM_ANY
                ^^^
	help
	  Range 3200 to 65534 is invalid. 65535 represents no specific value.
                      ^^^^
```

Same for `BT_PERIPHERAL_PREF_MIN_INT`
https://github.com/zephyrproject-rtos/zephyr/blob/df5318479b914d320f52870c6dfd0420779fc5c0/subsys/bluetooth/host/Kconfig.gatt#L231

jhedberg, 2025-03-10:
@robertinant I don't see the problem here. The documentation is talking about an *invalid* range, which is different for the Kconfig option range. It's not possible to represent multiple ranges using the Kconfig `range` directive.

robertinant, 2025-03-10:
I completely misread that. The key being INVALID. So no issue here.
With that said, trying to understand the choice of invalide vs valid. Why specify an invalid range vs specify a valid range instead.

jhedberg, 2025-03-10:
The valid (as per spec) range is non-contiguous, so it can't be completely represented with the Kconfig `range` directive. I believe that's why the help text is highlighting that a part of the range indicated by the `range` directive is actually not valid. Feel free to propose some rewording of it, but I'll close this issue for now since I don't see any obvious thing we could fix.

closed 2025-03-10 by jhedberg