zephyrproject-rtos/zephyr #86825: Missing BLE and WIFI support with Arduino Opta

stefannilsson, 2025-03-09:
**Is your enhancement proposal related to a problem?*
Lacking support for the HW supported Wifi/BLE.

**Describe the solution you'd like**
Provide Bluetooth support via the standard BLE APIs, as well as the Wifi APIs.

**Additional context**
Murata LBEE5KL1DX-883 radio module is used according to Declaration of Conformity docs.

I have one Arduino Opta AFX00002 available for use in testing.

[image]

henrikbrixandersen, 2025-03-09:
> I have one Arduino Opta AFX00002 available for use in testing.

Great! Looking forward to seeing a pull request for this.

stefannilsson, 2025-03-09:
The same BLE/Wifi chipset is apparently used in the Arduino Portenta H7 product.

However, this [Zephyr board](https://github.com/zephyrproject-rtos/zephyr/tree/df5318479b914d320f52870c6dfd0420779fc5c0/boards/arduino/portenta_h7) only seems to have BLE enabled atm. No Wifi support.

Anyone with experience with this chipset family and a possible understanding of the complexity of getting both Wifi and BLE enabled?

fabiobaltieri, 2025-03-10:
> Anyone with experience with this chipset family and a possible understanding of the complexity of getting both Wifi and BLE enabled?

Yeah, ble uses the standard transport over uart so it should work fine one you define the device. Wifi instead is connected over SDIO, and at the moment there's no SDIO driver for the STM32 SDIO controller. Vaguely remember the ST folks saying that they have that driver in their roadmap from when the airoc driver was introduced, but I'm not sure if they are actually actively working on it. The chip would also work under SPI but unfortunately the pins used are only mapped to the sdio interface. I had a go at trying to use it with a bitbanged spi interface recently but could not get the controller to load the firnware for some reason. So yeah this could use some help.

beriberikix, 2025-05-23:
I belive https://github.com/zephyrproject-rtos/zephyr/pull/89776 is the relevant issue for SDIO support.

beriberikix, 2025-07-30:
Note SDIO support is now merged onto `main`.

pillo79, 2025-07-30:
> SDIO support is ~now~finally merged onto main.

FTFY 🤣 On a more serious note, will work on using that on Arduino H7 boards (Portenta H7, Giga R1, Opta, Nicla Vision) after the holidays.

beriberikix, 2025-07-30:
Looking forward to it - I have most of those boards in a drawer waiting to connect!