zephyrproject-rtos/zephyr #86827: IPPROTO_RAW wrongly used with AF_PACKET

go2sh, 2025-03-09:
**Describe the bug**
Zephyr specifies the use of `IPPROTO_RAW` with `AF_PACKET` sockets. This is not in line how Linux specifies this. `IPPROTO_RAW´ is specified as sub type for IP frames. [[1]](https://www.iana.org/assignments/protocol-numbers/protocol-numbers.xhtml#protocol-numbers-1) Used as EtherType this specifies an old length field type. [[2]](https://www.iana.org/assignments/ieee-802-numbers/ieee-802-numbers.xhtml#ieee-802-numbers-1)

**To Reproduce**
Use AF_PACKET socket.

**Expected behavior**
`IPPROTO_RAW` is not treated as a protocol with `AF_PACKET` sockets, but with AF_INET/AF_INET6 through a SOCK_RAW socket.

**Impact**
Incompatibility (and maintaining this). overhead when using AF_PACKET.

**Logs and console output**
None.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: df5318479b914d320f52870c6dfd0420779fc5c0

**Additional context**
None.

jukkar, 2025-03-10:
The original plan was to be aligned with Linux here, but it seems that target was missed then. I suppose you are talking about this line https://github.com/zephyrproject-rtos/zephyr/blob/ab7a6de5bbb2a63e30390594363c4c3ec301b042/subsys/net/lib/sockets/sockets_packet.c#L56
I think we can easily remove that one, the original plan was to have some default value if user gives `proto` as 0 which is not a valid value.

go2sh, 2025-03-10:
It's not only this line. This adds additional overhead:
https://github.com/zephyrproject-rtos/zephyr/blob/658f9357b87ab3e02351a57183743ec70f92cdd7/subsys/net/ip/net_core.c#L131

go2sh, 2025-03-10:
> More info. According to this page https://stackoverflow.com/questions/49309029/confusion-with-af-inet-with-sock-raw-as-the-socket-type-v-s-af-packet-with-so the protocol field should be allowed to be 0 with `AF_PACKET`.

Here is also more info: https://man7.org/linux/man-pages/man7/raw.7.html
TLDR:
`IPPROTO_RAW` means  send only sockets and the IP header needs to be provided by the user.

jukkar, 2025-03-10:
I think raw7 manual page only talks about AF_INET/6 sockets, not AF_PACKET ones.

go2sh, 2025-03-10:
https://man7.org/linux/man-pages/man7/packet.7.html Doesn't talk about IPPROTO_RAW either.

AF_PACKET is a layer 2 socket and IPPROTO_RAW is used at l3/l4 sockets.

jukkar, 2025-03-10:
> Doesn't talk about IPPROTO_RAW either.

Yes, I noticed the same, it does not talk about IPPROTO_RAW. Only place IPPROTO_RAW is mentioned is in raw(7).

mkschreder, 2025-03-10:
See if my MR [86869](https://github.com/zephyrproject-rtos/zephyr/pull/86869) aligns with your use case.

jukkar, 2025-03-10:
I have alternative proposal at #86885, it tries to handle the case where protocol is set to 0. It also removes the `IPPROTO_RAW` from `AF_PACKET` as that was a wrong things to do.

go2sh, 2025-03-11:
> [@go2sh](https://github.com/go2sh) I constructed a table above to cover all combinations so that we can agree what would need to happen in different case. Please take a look if it makes sense to you.
> 
> Lets agree on the table first and then just implement it in a PR to get this working properly. We probably need to mention this table somewhere in the network documentation and link to that in the migration guide to help user to understand what is going on.

I had exactly the same on my mind. Looks good at first glance. I try to review it tomorrow in depth. I'm currently at embedded world in Nürnberg.

go2sh, 2025-03-13:
Two Remarks:
`IPPROTO_IP`: I checked the kernel code. https://elixir.bootlin.com/linux/v6.11.7/source/net/ipv4/af_inet.c#L277 For non-sock SOCK_RAW sockets, it selects the default protocol, for SOCK_RAW the `socket(...)` fails with protocol not supported.
Include the binding behavior: For AF_PACKET. Sockets without binding receive packets, which match the ethertype specified with `socket(...)`. Sockets with binding  receive packets from the interface and with the protocol specified during `bind(...)`
For AF_INET+SOCK_RAW: All packets for a protocol are received with out binding, packets are received for a local address with a binding.

rlubos, 2025-03-14:
> IPPROTO_IP: I checked the kernel code. https://elixir.bootlin.com/linux/v6.11.7/source/net/ipv4/af_inet.c#L277 For non-sock SOCK_RAW sockets, it selects the default protocol, for SOCK_RAW the socket(...) fails with protocol not supported.

About this behavior (failing on IPPROTO_IP) - there's a nice comparison in the article linked by Jukka: https://sock-raw.org/papers/sock_raw (see section 0x6. Summary), this doesn't seem to be standardized. FreeBSD on the other hand opens a socket that just forwards all IP packets on input, I'd rather lean towards this behavior, otherwise AF_INET(6)/SOCK_RAW sockets would be limited to a single L4 protocol only.

go2sh, 2025-03-19:
> I'd rather lean towards this behavior, otherwise AF_INET(6)/SOCK_RAW sockets would be limited to a single L4 protocol only.

I agree that this behavior makes more sense. I was just referring to the comment from @jukkar to be aligned with the Linux behavior.

jukkar, 2025-03-21:
I updated the table according to latest comments. Please take a look and if you notice issues, please adjust the table directly.

closed 2025-04-09 by kartben